boblemaire/IoTaWatt #54: No mesurement restart after disconnect/reconnect of VT

promy, 2017-08-26:
Using: 
- hardware version 4.2 (modified to through-hole and 3 VT channels)
- firmware version 2.0.15 (compiled from source only change is disabled auto-update)
Problem:
When disconnection the VT and reconnection it afterwards (or starting without the VT connected and plugging it in after boot) the measurements do not (re)start.
Only know solution is to cut the usb power.

I tried this 12 times in a row and 12 fails.
The first 10times i waited until VT reading hit 0 before reconnection. last 2 times i unplugged it for 2 - 5 seconds (before VT reading hit 0).

boblemaire, 2017-09-12:
@promy Is this still a problem or did it go away with the better USB power?

promy, 2017-09-12:
Sorry. Did forget to close this. It was solved with a more powerfull usb power. I havent had it again.

closed 2017-09-12 by promy