boblemaire/IoTaWatt #59: Restart Loop after shutdown/unplug Exception(3)

promy, 2017-09-13:
I unplugged my homebrew IotaWatt (unplug usb + 1 of the VT's).
Then i reconnected them, but with a splitter for the unplugged VT (connected to a second IotaWatt).
When rebooting the first IotaWatt kept flashing a red led.
On the usb-serial it stated a restart Exception (3). (see below).
However, putting in the sd card from the second one (a fresh copy) it booted without any problems. The second IotaWatt also gets in a restart loop with the sd card from the first.
Can it be that there was something corrupted? If so can this be prevented? (I unplugged it like 50times before without any problems. Only change was that now it was logging to emoncms.

The software is v2.02.15

The serial printout:
-------------------
** Restart **

Serial Initialized
No clock yet: SPI started.
No clock yet: SD initialized.
9/13/17 16:10:22 Real Time Clock is running. Unix time: 1505319022
9/13/17 16:10:22 Version: 2.02.15
9/13/17 16:10:22 Reset reason: Exception
9/13/17 16:10:22 Trace: 22128,37663,65378,32126,58124,57973,65413,44090,57159,48009,58697,6430,43023,45438,62294,649,59239,15433,15082,48132,35685,58884,26926,47677,44765,34633,47799,34911,38189,42638,62136,37845
9/13/17 16:10:22 ESP8266 ChipID:763523

Exception (3):
epc1=0x401003e9 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4000dfa1 depc=0x00000000

ctx: cont 
sp: 3fff39b0 end: 3fff3d80 offset: 01a0

>>>stack>>>
3fff3b50:  00000010 3fff3bf0 3fff3bf0 3fff2c78  
3fff3b60:  00000010 3fff3bf0 00000000 00000008  
3fff3b70:  00000000 3fff6ef8 3fff3bf0 401004d8  
3fff3b80:  3fff5401 00000000 3fff3bf0 40217c24  
3fff3b90:  00000010 3fff3bf0 3fff3bf0 40203b90  
3fff3ba0:  3fff6ed4 00000008 3fff3bf0 40217033  
3fff3bb0:  3fff2c78 00000005 3fff3bf0 40217065  
3fff3bc0:  3fff1ea8 00000000 3fff3bf0 4021716c  
3fff3bd0:  3ffe8c48 00000000 3fff6eec 3fff53f4  
3fff3be0:  3fff1ea8 00000000 3fff6eec 402074a7  
3fff3bf0:  3fff45a4 0000000f 00000008 3fff6ed4  
3fff3c00:  0000000f 00000008 3fff6ebc 0000000f  
3fff3c10:  00000008 00000000 00000000 3ff00000  
3fff3c20:  3fff5fec 3fff6ef8 3fff53e0 40203554  
3fff3c30:  3fff5668 3ffe8ce4 3ffe8cf4 0000000f  
3fff3c40:  3fff5668 3ffe8ce4 3ffe8cf4 40207d47  
3fff3c50:  3fff5668 3ffe8ce4 00000000 0000000a  
3fff3c60:  666e6f63 742e6769 00007478 00000000  
3fff3c70:  3fff4b34 3ffe9ce8 00000000 000003e8  
3fff3c80:  00000000 666e6f63 742e6769 00007478  
3fff3c90:  00000000 00000000 3ffe9600 3fff160a  
3fff3ca0:  3fff5d64 00001000 3fff5668 3ffe8ce4  
3fff3cb0:  00000000 00000015 00000015 4010020c  
3fff3cc0:  3fff464c 0000000f 0000000a 4010068c  
3fff3cd0:  3fff5640 3fff4ec4 3fff4b34 3fff6e2c  
3fff3ce0:  0000000e 3fff6e20 3ffe8f5c 402063ae  
3fff3cf0:  00000000 00000000 00000000 3fff25e4  
3fff3d00:  3fff2c7c 00000001 3fff2ab8 40208ffc  
3fff3d10:  00000000 00000000 00000000 feefeffe  
3fff3d20:  feefeffe feefeffe feefeffe feefeffe  
3fff3d30:  feefeffe feefeffe feefeffe feefeffe  
3fff3d40:  feefeffe feefeffe feefeffe feefeffe  
3fff3d50:  feefeffe feefeffe feefeffe 3fff2d58  
3fff3d60:  3fffdad0 00000000 3fff2d51 40217d44  
3fff3d70:  feefeffe feefeffe 3fff2d60 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000318
~ld
…
------------------------------------------

promy, 2017-09-13:
update: 
it seems that 
- deleting everything in the IotaWatt dir solved the problem
- unplugging it again recreated the problem? COuld the problem be in unplugging it while updating emoncms?

boblemaire, 2017-09-15:
There are a lot of moving parts to this issue.  There is a new VT splitter.  There are two IotaWatt devices.  There is a sequence of events that contains what you think is relevant but doesn't answer a bunch of questions that I would have in taking a methodical approach to the problem.  But the underlying question that you ask is could the problem be in unplugging it while updating Emoncms?

The answer is sure it could.  But more importantly, removing and reinserting the SDcard can and DOES cause corruption of the SDcard.  It's not meant to be removed.  If it is, the device should be powered off before reinserting.  There are no GPIO pins available to sense when the card is removed, so IotaWatt does not know it needs to close files and reinitialize the SD.  There is an operating system on the SDcard that is keeping track of unused blocks, and the context of the file system in the SD class instance has no idea that files have been modified externally.

I'm going to close this as it lacks information sufficient to recreate and involves procedures that are outside of the designed operation of the device (removal of the SD).  By all means, if you can resolve these issues, feel free to ask again.

closed 2017-09-15 by boblemaire