boblemaire/IoTaWatt #75: Share IoTaWatt schematics

skabashnyuk, 2017-10-09:
It would be nice if you can share eagle or gerber schematics of the board

flywire, 2017-10-09:
I think it is still this one:
https://github.com/boblemaire/IoTaWatt/blob/c0bb95f096adf0cbdddb9c2e1ff9c784bcd33448/Hardware/IotaWatt4.3.sch

IotaWatt
Electric Power Monitor
Version 4.1
July 6, 2017

Picture of PCB is labelled:
IoTaWatt
Rev 4.3

skabashnyuk, 2017-10-09:
Awesome. Would you like me to create a PR to add this file in https://github.com/boblemaire/IoTaWatt/tree/master/Hardware folder?  
BTW I can maintain Gerber file too.

skabashnyuk, 2017-10-09:
@flywire Silly question. What about IotaWatt4.3.brd file? Can you share it too?

boblemaire, 2017-10-09:
I'm looking into issues surrounding licensing and managing the feedback that comes from independent efforts to build this.  While it may seem a simple matter to just post the .brd file, to be useful the parts libraries, BOM etc.  need to be included.  Right now I am overwhelmed with software issues related to supporting the initial OEM offering.

skabashnyuk, 2017-10-09:
I can help with bom if you want to. For instance we can start with: 
```
Qty Value                Device               Package        Parts                                                               Description                                                                              
4   .1uf                 C-USC0805            C0805          C3, C4, C5, C6                                                      CAPACITOR, American symbol                                                               
2   10uF                 C-USC0805            C0805          C7, C9                                                              CAPACITOR, American symbol                                                               
2   10uf                 C-USC0805            C0805          C1, C2                                                              CAPACITOR, American symbol                                                               
1   12K                  R-US_R0805           R0805          R32                                                                 RESISTOR, American symbol                                                                
3   1K                   R-US_R0805           R0805          R3, R29, R31                                                        RESISTOR, American symbol                                                                
1   2.1MMJACKSMT         2.1MMJACKSMT         DCJACK_2MM_SMT J0                                                                  2.1mm x 5.5mm THM DC jack with internal switch. Digikey part #PJ-102A, 4UCON part #05537 
14  24                   R-US_R0805           R0805          R8, R12, R16, R20, R24, R28, R36, R40, R44, R48, R52, R56, R57, R58 RESISTOR, American symbol                                                                
1   32Khz                CRYSTAL8X3.8SMT      CRYSTAL_8X3.8  Y1                                                                  Crystals                                                                                 
2   4.7K                 R-US_R0805           R0805          R1, R2                                                              RESISTOR, American symbol                                                                
1   470                  R-US_R0805           R0805          R4                                                                  RESISTOR, American symbol                                                                
1   CR1220SMD            CR1220SMD            CR1220SMD      U$18                                                                CR1220 Battery SMD                                                                       
2   LED_RT_TRH           LED_RT_TRH           LED_TH_RT      U$15, U$23                                                          LED RT ThruHole BiColor Digikey 67-2237-ND                                               
1   LM358_DUAL_OP-AMP    LM358_DUAL_OP-AMP    SOIC_8         U$22                                                                LM358 Dual Op-Amp - SOIC8                                                                
2   MCP3208              MCP3208              SOIC_16        MCP3208-0, MCP3208-1                                                MCP3208 12bit SPI ADC                                                                    
1   MICROSD_PUSH/PULL    MICROSD_PUSH/PULL    MICROSD        MICROSD                                                             McroSD Amphenol push/pull                                                                
1   NODEMCU              NODEMCU              NODEMCU        U$16                                                                NodeMCU 1.0 ESP8266                                                                      
1   PCF8523_RTC          PCF8523_RTC          SOIC_8         PCF8523                                                             PCF8523-RTC SOIC-8                                                                       
14  STEREO_SMD           STEREO_SMD           STEREO_SMT     J1, J2, J3, J4, J5, J6, J7, J8, J9, J10, J11, J12, J13, J14         3.5mm Stereo Jack SMT                                                                    
4   TVS_4CHAN_ARRAYSC-74 TVS_4CHAN_ARRAYSC-74 SC-74          U$17, U$19, U$20, U$21                                              TVS 4 channel array                                                                      
1   VREF_SHUNT           VREF_SHUNT           SOT23_3        ADR510                                                              Voltage Reference Shunt - SOT23-3 (ADR504, etc.)                                         
```
or match it with https://octopart.com/ database. What is your vision for how it should looks like?

skabashnyuk, 2017-10-09:
@flywire @boblemaire direct question guys. Actually related to https://github.com/boblemaire/IoTaWatt/issues/74 : are you going to allow third-party contributors to manufacturing your board according to your PCB design? For home use? For commercial use? Or it's more proprietary product?

boblemaire, 2017-10-09:
Right now I am overwhelmed with software issues related to supporting the initial OEM offering.

flywire, 2017-10-10:
Good to see ‘soft’ community launch today.

On 18 September 2017 this schematic was posted:
https://github.com/boblemaire/IoTaWatt/blob/952609b34bc0bb3d88239f9fcf429b88e060000f/Hardware/IotaWatt4.3.sch
IotaWatt
Electric Power Monitor
Version 4.3
August 14, 2017

The schematic_4.3.png loaded the following day has the text 
IotaWatt
Electric Power Monitor
Version 4.1
July 6, 2017
I suspect Version 4.1 was incorrectly loaded as schematic_4.3.png

glynhudson, 2017-10-10:
> I suspect Version 4.1 was incorrectly loaded as schematic_4.3.png

Thanks for letting us know, I've created a PR to fix this. I think the current schematic is actually correct just incorrectly labelled. Anyway it will be corrected once PR is merged. 

https://github.com/boblemaire/IoTaWatt/pull/77

CAD files are being prepared will be shared under open-source licence very soon.

boblemaire, 2017-10-17:
Eagle files have been added.  The license speaks for itself.

closed 2017-10-17 by boblemaire