boblemaire/IoTaWatt #76: Add SCT006-000

glynhudson, 2017-10-09:
Would it be possible to add SCT006-000 to the Iotawat CT list, currently SCT013 and SCT019 are listed but not yet SCT006

[image]

Here's a comparison table 

[image]

boblemaire, 2017-10-09:
I just checked and it's in the master branch.  Can you verify?

glynhudson, 2017-10-10:
Yes, I can confirm. Thanks 👍 

I've just made a minor change to enter the full name of SCT006 > SCT006-000 to avoid any confusion and keep consistency with the other CT's. This change has been included in my PR https://github.com/boblemaire/IoTaWatt/pull/77 

https://github.com/boblemaire/IoTaWatt/pull/77/commits/9a6d381b335c2a790f462a6fbffe729b4263d77f

boblemaire, 2017-10-12:
PR #77 merged

closed 2017-10-12 by boblemaire

fredrike, 2018-01-18:
I've got a couple of SCT006 that is rated 20A: 25mA which corresponds to: http://en.yhdc.com/product/SCT006-612.html 

Will the SCT006-000 work ok or should I use a generic one (seems to be 1:800 turns on my version too)?

boblemaire, 2018-01-18:
That looks like the same one I tested.   You should be able to configure with the standard SCT006-000 menu pick.