boblemaire/IoTaWatt #88: Lost WiFi signal when first setup

rosspeel, 2017-11-01:
Got my new iotawatt today finally pluged it in, searched for the wifi access point of the device, found it, when going from choosing which access point for it to log into the internet, it didnt go any further, the lights are still red, green, green but no wifi signal is found at all now, power cycled it a few times with still the same red, green, green but no WiFi signal, i removed the cover and removed the SD card and reinserted it but still the same, i removed the sd card and checked to see if there was a file system in my windows PC and has a home folder called ".Trash-1000", and files called "cnfstyle.css, config.txt, edit.htm, graph.htm, index.htm, tables.txt.

https://www.dropbox.com/sh/6lpkx0oh95yfrrs/AAAAfzNhLjzpkWHLTDXytX0Sa?dl=0

This is the link for the file directory

boblemaire, 2017-11-01:
The SD card director looks OK.  The LED indicates that it is in AP mode waiting for you to connect to SSID Iota+(ESPdeviceid).  Some devices scan for available SSIDs more frequently than others.  I usually use an IOS device which works very well.

When it hasn't connected to WiFi at startup, the device initially stays in AP mode for up to 5 minutes.  Because your real-time clock is not yet set (from internet) in will remain continue to enter AP mode for 6 minute periods.  Please try using a different WiFi device to configure.

When you say you chose which access-point initially, are you saying you chose the Iotannnn AP to configure of that you successfully connected to the IotaWatt and chose your WiFi network from the list it presented?  I'm assuming it was the former.   If so, did you enter the new device startup password IotaWatt (case sensitive)?

When you do successfully connect to the IotaWatt AP, it uses what is called a "captive portal" to present the WiFi configuration page.  Some devices don't automatically make this happen, so once connected, you need to go to your browser and browse to 192.168.4.1.  See [this](https://guide.openenergymonitor.org/setup/iotawatt/#connect-wifi) for an explanation of that dialogue.

rosspeel, 2017-11-02:
I did get to the WiFi managers page and selected my AP, that's where i got up to yesterday, today after leaving it on for a long while while doing other things i came back and the light had gone green and was not able to find it on my network under url: iotawatt.local. I then used a Ip scan tool and found it under 192.168.1.61 and logged on to it with that IP address, i was able to go through the menus and select each channels settings from 1 to 14 and set the voltage to my grid voltage, I have taken a few screenshots of this.
But since then i have set this my on my kitchen bench and after changing power points for the powering of the device i have now unable to log in to the device but i can ping it and i now have 3 flashing lights red - red - red. when i power cycle the device goes offline while watching the pinging and after power cycle startup quickly re pings through command prompt -t. The connection through chrome now refused to connect, even though i can ping it.

3 Flashing red lights video
https://www.dropbox.com/s/1ld8g4sgwt9rj2c/3%20Flashing%20Red%20Lights.mp4?dl=0

[image]
[image]
[image]
[image]
[image]
[image]