boblemaire/IoTaWatt #90: How do you configure an input for a load that uses both phases but only a single CT

Frogmore42, 2017-11-02:
In the US we have 240 split phase power.  For 240V only loads (pumps, water heater, AC compressor would be examples) only a single CT is needed, since the neutral is not used.  This means that the actual power/energy used is twice what is measured.  I found the Outputs screen, which lets me create something that represents the total power by multiplying the measure power by 2.  But, I did not see a way to do this for the energy.

boblemaire, 2017-11-02:
They are the same, but the graph app does not list the outputs under the energy tab. I have added that but the changes are not yet merged into a release.  I’ll put that out soon and mark this closed when I do.

As an aside, I too am on the US power system, and when I can, I resolve those 240 connections by passing both leads through the CT, one in the opposite direction.  This works well on those devices that use the neutral for 120 volt, and also doubles the current, which gives it more resolution at low values.  It requires a CT rated for double the current the current the device uses, and sometimes the box and wire length won’t accommodate, but it’s a consideration.  What you have done is fine as well.
 One that if you upload to EmonCMS or another server, you can use the calculator to double the value there as well, so you won’t need to do that in the input process list.

Frogmore42, 2017-11-03:
I found another solution too.  I just added a new entry to tables.txt that was the same CT with a x2 added to the name and then doubled the turns value.  That appears to double the output value and everything works normally without needing to create another feed.

boblemaire, 2017-11-03:
If that works for you.  I don’t recommend to anyone else relying on modifications to the table file.  The table file is distributed with each new release and user changes will be gone.

closed 2017-11-03 by boblemaire