boblemaire/IoTaWatt #105: Feature request:  Add Help Page to each tab

JonMurphy, 2017-11-22:
Add a help page link [image] with link to wiki for each tab.
[image]
This "circle ?" in the bottom right corner would point to:
https://github.com/boblemaire/IoTaWatt/wiki/Device-Configuration

Icon Link: https://www.flaticon.com/free-icon/question-mark-inside-a-circle_827

PS - I know your TO-DO list is long.  I'd be happy to help.  If you show me how to do one tab I can figure out the rest and do the needed push requests.

boblemaire, 2018-01-15:
Jon,
I'm interested in doing this, but I'd like to avoid linking to an external image for the icon.  Do you think you can source an embedded base64 image to use?

JonMurphy, 2018-01-17:
I think the link I sent had the ability to do a embedded base64 image (without linking an external image).  I'll read thru it again.

glynhudson, 2018-01-17:
Good idea Jon

- sent from my mobile device

JonMurphy, 2018-01-17:
To me this doesn't look linked.  Here is the Base 64 encoded image (HTML):

```
<img src="data:image/svg+xml;utf8;base64,PD94bWwgdmVyc2lvbj0iMS4wIiBlbmNvZGluZz0iaXNvLTg4NTktMSI/Pgo8IS0tIEdlbmVyYXRvcjogQWRvYmUgSWxsdXN0cmF0b3IgMTguMS4xLCBTVkcgRXhwb3J0IFBsdWctSW4gLiBTVkcgVmVyc2lvbjogNi4wMCBCdWlsZCAwKSAgLS0+CjxzdmcgeG1sbnM9Imh0dHA6Ly93d3cudzMub3JnLzIwMDAvc3ZnIiB4bWxuczp4bGluaz0iaHR0cDovL3d3dy53My5vcmcvMTk5OS94bGluayIgdmVyc2lvbj0iMS4xIiBpZD0iQ2FwYV8xIiB4PSIwcHgiIHk9IjBweCIgdmlld0JveD0iMCAwIDI5LjUzNiAyOS41MzYiIHN0eWxlPSJlbmFibGUtYmFja2dyb3VuZDpuZXcgMCAwIDI5LjUzNiAyOS41MzY7IiB4bWw6c3BhY2U9InByZXNlcnZlIiB3aWR0aD0iNTEycHgiIGhlaWdodD0iNTEycHgiPgo8Zz4KCTxwYXRoIGQ9Ik0xNC43NjgsMEM2LjYxMSwwLDAsNi42MDksMCwxNC43NjhjMCw4LjE1NSw2LjYxMSwxNC43NjcsMTQuNzY4LDE0Ljc2N3MxNC43NjgtNi42MTIsMTQuNzY4LTE0Ljc2NyAgIEMyOS41MzUsNi42MDksMjIuOTI0LDAsMTQuNzY4LDB6IE0xNC43NjgsMjcuMTI2Yy02LjgyOCwwLTEyLjM2MS01LjUzMi0xMi4zNjEtMTIuMzU5YzAtNi44MjgsNS41MzMtMTIuMzYyLDEyLjM2MS0xMi4zNjIgICBjNi44MjYsMCwxMi4zNTksNS41MzUsMTIuMzU5LDEyLjM2MkMyNy4xMjcsMjEuNTk0LDIxLjU5NCwyNy4xMjYsMTQuNzY4LDI3LjEyNnoiIGZpbGw9IiMwMDAwMDAiLz4KCTxwYXRoIGQ9Ik0xNC4zODUsMTkuMzM3Yy0xLjMzOCwwLTIuMjg5LDAuOTUxLTIuMjg5LDIuMzRjMCwxLjMzNiwwLjkyNiwyLjMzOSwyLjI4OSwyLjMzOWMxLjQxNCwwLDIuMzE0LTEuMDAzLDIuMzE0LTIuMzM5ICAgQzE2LjY3MiwyMC4yODgsMTUuNzcxLDE5LjMzNywxNC4zODUsMTkuMzM3eiIgZmlsbD0iIzAwMDAwMCIvPgoJPHBhdGggZD0iTTE0Ljc0Miw2LjA5MmMtMS44MjQsMC0zLjM0LDAuNTEzLTQuMjkzLDEuMDUzbDAuODc1LDIuODA0YzAuNjY4LTAuNDYyLDEuNjk3LTAuNzcyLDIuNTQ1LTAuNzcyICAgYzEuMjg1LDAuMDI3LDEuODc5LDAuNjQ0LDEuODc5LDEuNTQzYzAsMC44NS0wLjY3LDEuNjk3LTEuNDk0LDIuNzAxYy0xLjE1NiwxLjM2NC0xLjU5NCwyLjcwMS0xLjUxNiw0LjAxMmwwLjAyNSwwLjY2OWgzLjQyICAgdi0wLjQ2M2MtMC4wMjUtMS4xNTgsMC4zODctMi4xNjIsMS4zMTEtMy4yMTVjMC45NzktMS4wOCwyLjIxMS0yLjM2NiwyLjIxMS00LjMyMUMxOS43MDUsNy45NjgsMTguMTM5LDYuMDkyLDE0Ljc0Miw2LjA5MnoiIGZpbGw9IiMwMDAwMDAiLz4KCTxnPgoJPC9nPgoJPGc+Cgk8L2c+Cgk8Zz4KCTwvZz4KCTxnPgoJPC9nPgoJPGc+Cgk8L2c+Cgk8Zz4KCTwvZz4KCTxnPgoJPC9nPgoJPGc+Cgk8L2c+Cgk8Zz4KCTwvZz4KCTxnPgoJPC9nPgoJPGc+Cgk8L2c+Cgk8Zz4KCTwvZz4KCTxnPgoJPC9nPgoJPGc+Cgk8L2c+Cgk8Zz4KCTwvZz4KPC9nPgo8Zz4KPC9nPgo8Zz4KPC9nPgo8Zz4KPC9nPgo8Zz4KPC9nPgo8Zz4KPC9nPgo8Zz4KPC9nPgo8Zz4KPC9nPgo8Zz4KPC9nPgo8Zz4KPC9nPgo8Zz4KPC9nPgo8Zz4KPC9nPgo8Zz4KPC9nPgo8Zz4KPC9nPgo8Zz4KPC9nPgo8Zz4KPC9nPgo8L3N2Zz4K" />
```

From the License it looks like the author is looking for a mention.  Maybe this could be your **About** page or a **Credits** page.
> Icon made by Freepik from www.flaticon.com 

There is a license PDF on the site.  You'd have to answer if it meets their terms & conditions.

boblemaire, 2018-01-17:
Yeah, I looked at it.  That's why I asked if you would source the symbol.  I'm not going to add code to comply with the copyright for a question mark in a circle.  I don't blame the guy for wanting it, I just don't think it's worth the trouble. I'd just as soon put a little button there with the work Wiki,  Probably that's what I'll do when I get around to it.

boblemaire, 2018-06-11:
With no activity on this for many months, I'm going to close it.  I have added HTML titles to many of the items in the menus, and also added some pretty long explanations in some of the menus.  The WiKi is going to be converted to read-the-docs soon.

closed 2018-06-11 by boblemaire