boblemaire/IoTaWatt #107: BOM/Part Library

warren-woolsey, 2017-12-05:
Would it be possible to get a BOM or Part Library to go along with the Eagle Files?

boblemaire, 2017-12-05:
I believe the part information contained in the Eagle files is accurate.  The only issue might be the VREF-shunt which is an sot23-3 2.5V .2% shunt, generally available from major suppliers.
It just increases my work and opens the door for more error if I maintain something separate from the Schematics.

warren-woolsey, 2017-12-06:
It would be simple to just drop a copy of the Library file for the board when you drop a copy of the schematic and board files. This would give us access to the list of the actual parts used, rather than just making a best guess from the descriptions.

bgupta55, 2017-12-06:
I agree. I am from India, having tough time to procure following components :

1. LED_RT_TRH - (LED RT ThruHole BiColor Digikey 67-2237-ND) - looks like only available on digikey, and shipping to India is expensive.

2. 3.5mm Stereo Jack SMT - looks like only available on digikey, and shipping to India is expensive.

3. TVS 4 channel array

I am not able to find these in local market, as well as on aliexpress. 

Request you to please suggest.

Thanks.

warren-woolsey, 2017-12-26:
I'm having issues with the Stereo Jacks my self.

namezero, 2018-01-01:
the Sereo jack reference is  CP-3524SJTR-ND right ? 
--

mattncsu, 2018-01-17:
I put together this BOM.  Has everything but the ESP. No guarantee it is accurate, complete, or the same parts @boblemaire used.  I just put it together to get a quick estimate of the cost to build an iotaWatt vs emontx/emontx shield/emonpi boards. https://octopart.com/bom-tool/C2JyOqfc

erice1984, 2019-01-16:
Tkfu marking on vref shunt which identifies as a TL4050.  I belive the tvs diodes are these D5V0P4UR6SO-7.

boblemaire, 2019-01-16:
TL4050B25IDBZR‎ but any SOT23-3 shunt 3V or less should work.  The firmware assumes 2.5V but can overide in the config and soon in the EEPROM at manufacture.  I use 0.2% tolerance.

closed 2019-01-16 by boblemaire