boblemaire/IoTaWatt #109: ESP32 update

gesaleh, 2017-12-10:
Hi
I would like to help on the ESP32 update
what is the status ? are you going to have WiFi or BLE active ? 
I have several ESP32 boards ready to be integrated contact me to see how we can work together if needed

Thanks

Onepamopa, 2017-12-10:
Should be option for both BLE or Wi-Fi, user's choice.

boblemaire, 2017-12-14:
It's a complete rewrite.  The opportunities that are presented with the ESP32 would be difficult to exploit effectively within the framework of the ES8266 version.  Specifically, I want to use the multi-processor resource optimally.  So it's looking like an enhanced scheduler with maybe the sampling happening on a dedicated processor.  It's all still very vague.  I'm trying to get some experience with it.

closed 2018-01-15 by boblemaire

Onepamopa, 2018-03-20:
So, are we gonna see an ESP32 version or we should forget about it?

boblemaire, 2018-03-23:
I would say it's a long way off.  There are two issues:
1) This ESP8266 version is taking off and I have a lot to do on it yet (that should be portable to an ESP version)
2) Last time I checked there was no TLS or SSL support in the asyncTCP for the ESP32.  I'm not using it in the ESP8266 either, but if I'm going to port, I'll go the extra distance to move to a platform that offers better security options.