boblemaire/IoTaWatt #111: Feeds missing from version of software

rosspeel, 2017-12-15:
Been away for a few days and went to check the graphs on my solar output and looks like a new version of software has been updated, but now it seems that my feeds has disappeared on my graphs
[image]

Frogmore42, 2017-12-15:
Refresh a few times and go back to the main page.  I did that a few times and the feeds magically showed up.

boblemaire, 2017-12-15:
The newer release includes updated versions of the apps, including the configuration and graph app.  To make sure you are running the latest versions, please close your browser windows and start fresh with the config app (iotawatt.local).  Then run the graph from there.  If you still do not have feeds, check the status page.  Make sure you the device is updating status.  Click on the Data Logs tab and post a screenshot of the status.

rosspeel, 2017-12-16:
Have rebooted and refresh many times from different PCs but still no graphs to be found, have taken screenshots of status page and graph page
[image]
[image]

boblemaire, 2017-12-16:
OK, that looks pretty busy.  Can you enter this url:

http://iotawatt.local/feed/list.json

and post the result?

boblemaire, 2017-12-16:
I think this has something to do with spaces in the output channel names.  I notice the calculator indicates such a name is invalid but allows SAVE.  That's something that must be fixed.  Also, your prior release probably allowed it and worked fine.  The API was changed to use the names rather than a numeric token so that it could be useful to other applications.
I see that you have a lot invested in naming a lot of outputs.  Could you try using underscores in all of them as in the input names?

rosspeel, 2017-12-16:
list.jason

[{"id":"IVVoltage","tag":"Voltage","name":"Voltage"},{"id":"IPImport_Export","tag":"Power","name":"Import_Export"},{"id":"IEImport_Export","tag":"Energy","name":"Import_Export"},{"id":"IPSolar_East_5000w","tag":"Power","name":"Solar_East_5000w"},{"id":"IESolar_East_5000w","tag":"Energy","name":"Solar_East_5000w"},{"id":"IPSolar_West_5000w","tag":"Power","name":"Solar_West_5000w"},{"id":"IESolar_West_5000w","tag":"Energy","name":"Solar_West_5000w"},{"id":"IPSolar_W_and_N_With_Bat_4900w","tag":"Power","name":"Solar_W_and_N_With_Bat_4900w"},{"id":"IESolar_W_and_N_With_Bat_4900w","tag":"Energy","name":"Solar_W_and_N_With_Bat_4900w"},{"id":"IPTotal_Solar_Produced","tag":"Power","name":"Total_Solar_Produced"},{"id":"IETotal_Solar_Produced","tag":"Energy","name":"Total_Solar_Produced"},{"id":"IPHot_Water_House","tag":"Power","name":"Hot_Water_House"},{"id":"IEHot_Water_House","tag":"Energy","name":"Hot_Water_House"},{"id":"IPHouse_Heating_Cooling","tag":"Power","name":"House_Heating_Cooling"},{"id":"IEHouse_Heating_Cooling","tag":"Energy","name":"House_Heating_Cooling"},{"id":"IPShed_Power_A","tag":"Power","name":"Shed_Power_A"},{"id":"IEShed_Power_A","tag":"Energy","name":"Shed_Power_A"},{"id":"IPShed_Power_B","tag":"Power","name":"Shed_Power_B"},{"id":"IEShed_Power"},{"id":"IEHouse_Outdoor_Power","tag":"Energy","name":"House_Outdoor_Power"},{"id":"IPShed_32Amp_Outlet","tag":"Power","name":"Shed_32Amp_Outlet"},{"id":"IEShed_32Amp_Outlet","tag":"Energy","name":"Shed_32Amp_Outlet"},{"id":"OPHeating and Cooling","tag":"Power","name":"Heating and Cooling"},{"id":"OEHeating and Cooling","tag":"Energy","name":"Heating and Cooling"},{"id":"OPHouse General","tag":"Power","name":"House General"},{"id":"OEHouse General","tag":"Energy","name":"House General"},{"id":"OPHouse Hot Water","tag":"Power","name":"House Hot Water"},{"id":"OEHouse Hot Water","tag":"Energy","name":"House Hot Water"},{"id":"OPImport Export","tag":"Power","name":"Import Export"},{"id":"OEImport Export","tag":"Energy","name":"Import Export"},{"id":"OVMains Voltage","tag":"Voltage","name":"Mains Voltage"},{"id":"OPNBN Outdoor Box","tag":"Power","name":"NBN Outdoor Box"},{"id":"OENBN Outdoor Box","tag":"Energy","name":"NBN Outdoor Box"},{"id":"OPSelf Use","tag":"Power","name":"Self Use"},{"id":"OESelf Use","tag":"Energy","name":"Self Use"},{"id":"OPShed 32 Amp Outlet","tag":"Power","name":"Shed 32 Amp Outlet"},{"id":"OEShed 32 Amp Outlet","tag":"Energy","name":"Shed 32 Amp Outlet"},{"id":"OPShed A Circuit","tag":"Power","name":"Shed A Circuit"},{"id":"OEShed A Circuit","tag":"Energy","name":"Shed A Circuit"},{"id":"OPShed B Circuit With Pump","tag":"Power","name":"Shed B Circuit With Pump"},{"id":"OEShed B Circuit With Pump","tag":"Energy","name":"Shed B Circuit With Pump"},{"id":"OPShed Lighting","tag":"Power","name":"Shed Lighting"},{"id":"OEShed Lighting","tag":"Energy","name":"Shed Lighting"},{"id":"OPShed Rear Box","tag":"Power","name":"Shed Rear Box"},{"id":"OEShed Rear Box","tag":"Energy","name":"Shed Rear Box"},{"id":"OPSolar Bulk Production","tag":"Power","name":"Solar Bulk Production"},{"id":"OESolar Bulk Production","tag":"Energy","name":"Solar Bulk Production"},{"id":"OPSolar East Production","tag":"Power","name":"Solar East Production"},{"id":"OESolar East Production","tag":"Energy","name":"Solar East Production"},{"id":"OPSolar North Production","tag":"Power","name":"Solar North Production"},{"id":"OESolar North Production","tag":"Energy","name":"Solar North Production"},{"id":"OPSolar West Production","tag":"Power","name":"Solar West Production"},{"id":"OESolar West Production","tag":"Energy","name":"Solar West Production"}]

rosspeel, 2017-12-16:
I see that you have a lot invested in naming a lot of outputs. Could you try using underscores in all of them as in the input names?

Yes i have been using underscores in names
[image]

rosspeel, 2017-12-16:
message log

12/16/17 10:29:11 MDNS responder started
12/16/17 10:29:11 You can now connect to http://IotaWatt.local
12/16/17 10:29:11 HTTP server started
12/16/17 10:29:12 dataLog: service started.
12/16/17 10:29:12 dataLog: Last log entry:1513384145
12/16/17 10:29:12 statService: started.
12/16/17 10:29:12 timeSync: service started.
12/16/17 10:29:12 WiFi connected. SSID: Rosshouse AP, IP: 192.168.50.102
12/16/17 10:29:13 historyLog: service started.
12/16/17 10:29:13 historyLog: Last log entry:1513384140
12/16/17 10:29:17 EmonService: started.url: emoncms.org:80, node: Peel, post interval: 10, encrypted POST
12/16/17 10:29:17 EmonService: Start posting at 1513265360
12/16/17 10:29:26 EmonService: POST failed. HTTP code:read Timeout
12/16/17 10:30:26 EmonService: Resending EmonCMS data:2
12/16/17 10:32:27 EmonService: Resending EmonCMS data:3
12/16/17 10:35:28 EmonService: Resending EmonCMS data:4
12/16/17 10:39:30 EmonService: Resending EmonCMS data:5
12/16/17 10:44:32 EmonService: Resending EmonCMS data:6
12/16/17 10:50:33 EmonService: Resending EmonCMS data:7
12/16/17 10:57:34 EmonService: Resending EmonCMS data:8
12/16/17 11:05:35 EmonService: Resending EmonCMS data:9
12/16/17 11:14:37 EmonService: Resending EmonCMS data:10
12/16/17 11:14:38 EmonService: Unable to post to Emoncms after 10 retries.  Restarting ESP.

** Restart **

No clock yet: SD initialized.
12/16/17 01:14:39 Real Time Clock is running. Unix time: 1513386879
12/16/17 01:14:39 Version: 02_02_27
12/16/17 01:14:39 Reset reason: Software/System restart
12/16/17 01:14:39 Trace: 99,12,13,14,15,16,17,18,13,14,15,16,17,18,13,14,15,30,37,38,148,149,140,142,143,144,145,148,149,146,147,39
12/16/17 01:14:39 ESP8266 ChipID:416947
12/16/17 11:14:39 device name: IotaWatt, version: 3
12/16/17 11:14:39 Local time zone: 10
12/16/17 11:14:42 MDNS responder started
12/16/17 11:14:42 You can now connect to http://IotaWatt.local
12/16/17 11:14:42 HTTP server started
12/16/17 11:14:42 dataLog: service started.
12/16/17 11:14:43 dataLog: Last log entry:1513386875
12/16/17 11:14:43 statService: started.
12/16/17 11:14:43 timeSync: service started.
12/16/17 11:14:43 WiFi connected. SSID: Rosshouse AP, IP: 192.168.50.102
12/16/17 11:14:43 historyLog: service started.
12/16/17 11:14:44 historyLog: Last log entry:1513386840
12/16/17 11:14:47 EmonService: started.url: emoncms.org:80, node: Peel, post interval: 10, encrypted POST
12/16/17 11:14:47 EmonService: Start posting at 1513265360
12/16/17 11:14:55 EmonService: POST failed. HTTP code:read Timeout
12/16/17 11:15:23 Server: Sent less data than expected!
12/16/17 11:15:55 EmonService: Resending EmonCMS data:2
12/16/17 11:16:46 Server: Sent less data than expected!
12/16/17 11:17:56 EmonService: Resending EmonCMS data:3
12/16/17 11:20:43 Restart command received.

** Restart **

No clock yet: SD initialized.
12/16/17 01:20:44 Real Time Clock is running. Unix time: 1513387244
12/16/17 01:20:44 Version: 02_02_27
12/16/17 01:20:44 Reset reason: Software/System restart
12/16/17 01:20:44 Trace: 14,17,18,13,14,17,18,13,14,17,18,13,14,17,18,13,14,17,18,13,14,17,18,11,90,84,86,88,12,13,102,103
12/16/17 01:20:44 ESP8266 ChipID:416947
12/16/17 11:20:44 device name: IotaWatt, version: 3
12/16/17 11:20:44 Local time zone: 10
12/16/17 11:20:47 MDNS responder started
12/16/17 11:20:47 You can now connect to http://IotaWatt.local
12/16/17 11:20:47 HTTP server started
12/16/17 11:20:48 dataLog: service started.
12/16/17 11:20:48 dataLog: Last log entry:1513387240
[... 93 lines cut ...]
No clock yet: SD initialized.
12/16/17 03:38:44 Real Time Clock is running. Unix time: 1513395524
12/16/17 03:38:44 Version: 02_02_27
12/16/17 03:38:44 Reset reason: Software/System restart
12/16/17 03:38:44 Trace: 99,12,13,14,15,16,17,18,13,14,15,16,17,18,13,14,15,30,37,38,148,149,140,142,143,144,145,148,149,146,147,39
12/16/17 03:38:44 ESP8266 ChipID:416947
12/16/17 13:38:44 device name: IotaWatt, version: 3
12/16/17 13:38:44 Local time zone: 10
12/16/17 13:38:47 MDNS responder started
12/16/17 13:38:47 You can now connect to http://IotaWatt.local
12/16/17 13:38:47 HTTP server started
12/16/17 13:38:47 dataLog: service started.
12/16/17 13:38:48 dataLog: Last log entry:1513395520
12/16/17 13:38:48 statService: started.
12/16/17 13:38:48 timeSync: service started.
12/16/17 13:38:48 WiFi connected. SSID: Rosshouse AP, IP: 192.168.50.102
12/16/17 13:38:48 historyLog: service started.
12/16/17 13:38:49 historyLog: Last log entry:1513395480
12/16/17 13:38:52 EmonService: started.url: emoncms.org:80, node: Peel, post interval: 10, encrypted POST
12/16/17 13:38:52 EmonService: Start posting at 1513265360
12/16/17 13:39:00 EmonService: POST failed. HTTP code:read Timeout
12/16/17 13:39:36 Server: Sent less data than expected!
12/16/17 13:40:00 EmonService: Resending EmonCMS data:2
12/16/17 13:42:01 EmonService: Resending EmonCMS data:3
12/16/17 13:45:02 EmonService: Resending EmonCMS data:4

boblemaire, 2017-12-16:
I can see another problem here.  Your device has been unable to upload to Emoncms.org since Dec 14 3:29pm GMT.  

This will probably take some time to sort out.  The good news is that your IoTaWatt seems to be logging power fine and the data seems to be going to the datalog fine.  So it's a matter of getting to it from the graph utility and Emoncms.

I need to get a better context on your installation. Can you locate the portion of the datalog from Dec 14 3:29pm GMT.  You appear to be GMT +10 so Dec 15 1:29am.  When you view the datalog, it returns the last 10,000 characters.  To go further back, you can:

download the entire file by going to the file manager app, expand the +iotawatt directory in the left column, right click iotamsgs.txt and click download.

view the entire file in your browser with 
`192.168.50.102/iotawatt/iotamsgs.txt`

Also, could you post your config file using the same methods.  The file is config.txt in the root directory.
Posting the config, you may want to obfuscate your Emoncms apikey and account. 

After that,, as a diagnostic, could you remove the Emoncms userID (using the configuration app) and see if posting restarts.  This would use the unencrypted post.  There are ongoing problems with the latency of communicating with the UK from Australia.  I am working on changes to mitigate those problems.  That said, it could well be unrelated.

rosspeel, 2017-12-17:
Part of iotamsg.txt file.

12/13/17 15:51:51 Resending EmonCMS data.
12/13/17 16:03:21 EmonService: POST failed. HTTP code: read Timeout
12/13/17 16:03:51 Resending EmonCMS data.
12/13/17 16:14:53 EmonService: POST failed. HTTP code: read Timeout
12/13/17 16:15:23 Resending EmonCMS data.
12/13/17 16:51:04 EmonService: POST failed. HTTP code: read Timeout
12/13/17 16:51:34 Resending EmonCMS data.
12/13/17 17:03:03 EmonService: POST failed. HTTP code: send payload failed
12/13/17 17:03:33 Resending EmonCMS data.
12/13/17 17:05:04 timeSync: Failed to get NTPtime.
12/13/17 18:14:01 EmonService: POST failed. HTTP code: read Timeout
12/13/17 18:14:32 Resending EmonCMS data.
12/13/17 18:39:21 EmonService: POST failed. HTTP code: read Timeout
12/13/17 18:39:51 Resending EmonCMS data.
12/13/17 21:04:11 EmonService: POST failed. HTTP code: read Timeout
12/13/17 21:04:41 Resending EmonCMS data.
12/13/17 21:08:51 EmonService: POST failed. HTTP code: read Timeout
12/13/17 21:09:21 Resending EmonCMS data.
12/13/17 22:15:01 EmonService: POST failed. HTTP code: read Timeout
12/13/17 22:15:31 Resending EmonCMS data.
12/13/17 22:46:41 EmonService: POST failed. HTTP code: read Timeout
12/13/17 22:47:11 Resending EmonCMS data.
12/14/17 02:10:41 EmonService: POST failed. HTTP code: read Timeout
12/14/17 02:11:11 Resending EmonCMS data.
12/14/17 04:41:46 EmonService: POST failed. HTTP code: send header failed
12/14/17 04:42:16 Resending EmonCMS data.
12/14/17 08:08:19 checkUpdate: Invalid response from server.
12/14/17 08:22:02 EmonService: POST failed. HTTP code: send payload failed
12/14/17 08:22:32 Resending EmonCMS data.
12/14/17 08:43:11 EmonService: POST failed. HTTP code: read Timeout
12/14/17 08:43:41 Resending EmonCMS data.
12/14/17 09:10:41 EmonService: POST failed. HTTP code: read Timeout
12/14/17 09:11:11 Resending EmonCMS data.
12/14/17 09:22:45 EmonService: POST failed. HTTP code: send header failed
12/14/17 09:23:15 Resending EmonCMS data.
12/14/17 10:06:11 EmonService: POST failed. HTTP code: read Timeout
12/14/17 10:06:41 Resending EmonCMS data.
12/14/17 10:08:20 Updater: Update from 02_02_21 to 02_02_27
12/14/17 10:09:29 Updater: Update downloaded and signature verified
12/14/17 10:09:34 Updater: firmware upgraded to version 02_02_27
12/14/17 10:09:34 Firmware updated, restarting.

** Restart **

No clock yet: SD initialized.
12/14/17 00:09:42 Real Time Clock is running. Unix time: 1513210182
12/14/17 00:09:42 Version: 02_02_27
12/14/17 00:09:42 Updater: Installing update files for version 02_02_27
12/14/17 00:09:42 Updater: Installing GRAPH.JS
12/14/17 00:09:42 Updater: Installing INDEX.HTM
12/14/17 00:09:44 Updater: Installing TABLES.TXT
12/14/17 00:09:44 Updater: Installing CNFSTYLE.CSS
12/14/17 00:09:44 Updater: Installing EDIT.HTM
12/14/17 00:09:45 Updater: Installing GRAPH.HTM
12/14/17 00:09:45 Updater: Installation complete.
12/14/17 00:09:45 Reset reason: Software/System restart
12/14/17 00:09:45 Trace: 18,13,14,17,18,13,14,17,18,13,14,17,18,13,14,17,18,13,14,17,18,13,14,17,18,13,14,17,18,13,14,15
12/14/17 00:09:45 ESP8266 ChipID:416947
12/14/17 10:09:45 device name: IotaWatt, version: 3
12/14/17 10:09:45 Local time zone: 10
12/14/17 10:09:46 checkUpdate: Invalid response from server.
12/14/17 10:09:46 MDNS responder started
12/14/17 10:09:47 You can now connect to http://IotaWatt.local
12/14/17 10:09:47 HTTP server started
12/14/17 10:09:47 dataLog: service started.
12/14/17 10:09:47 dataLog: Last log entry:1513210095
12/14/17 10:09:47 statService: started.
12/14/17 10:09:47 timeSync: service started.
[... 1257 lines cut ...]
12/15/17 20:11:21 WiFi connected. SSID: Rosshouse AP, IP: 192.168.50.102
12/15/17 20:11:23 checkUpdate: Invalid response from server.
12/15/17 20:11:23 historyLog: service started.
12/15/17 20:11:23 historyLog: Last log entry:1513332660
12/15/17 20:11:26 EmonService: started.url: emoncms.org:80, node: Peel, post interval: 10, encrypted POST
12/15/17 20:11:26 EmonService: input/get failed.
12/15/17 20:11:57 EmonService: input/get failed.
12/15/17 20:12:28 EmonService: input/get failed.
12/15/17 20:12:59 EmonService: input/get failed.
12/15/17 20:13:30 EmonService: input/get failed.
12/15/17 20:14:01 EmonService: input/get failed.
12/15/17 20:14:32 EmonService: input/get failed.
12/15/17 20:15:03 EmonService: input/get failed.
12/15/17 20:15:34 EmonService: input/get failed.
12/15/17 20:16:05 EmonService: input/get failed.
12/15/17 20:16:26 timeSync: Failed to get NTPtime.
12/15/17 20:16:36 EmonService: input/get failed.
12/15/17 20:17:07 EmonService: input/get failed.
12/15/17 20:17:38 EmonService: input/get failed.
12/15/17 20:18:09 EmonService: input/get failed.
12/15/17 20:18:40 EmonService: input/get failed.
12/15/17 20:19:12 EmonService: input/get failed.
12/15/17 20:19:43 EmonService: input/get failed.
12/15/17 20:20:14 EmonService: input/get failed.
12/15/17 20:20:45 EmonService: input/get failed.

rosspeel, 2017-12-17:
config file

{
	"format": 2,
	"timezone": "10",
	"device": {
		"name": "IotaWatt",
		"version": 3,
		"channels": "15",
		"burden": [
			0,
			0,
			0,
			0,
			0,
			0,
			0,
			0,
			0,
			0,
			0,
			0,
			0,
			0,
			0
		]
	},
	"inputs": [
		{
			"channel": 0,
			"name": "Voltage",
			"type": "VT",
			"model": "generic240V",
			"phase": 2,
			"cal": 21.75
		},
		{
			"channel": 1,
			"name": "Import_Export",
			"type": "CT",
			"model": "generic",
			"phase": "2",
			"cal": "186",
			"signed": true
		},
		{
			"channel": 2,
			"name": "Solar_East_5000w",
			"type": "CT",
			"model": "generic",
			"phase": "2",
			"cal": "112"
		},
		{
			"channel": 3,
			"name": "Solar_West_5000w",
			"type": "CT",
			"model": "generic",
			"phase": "2",
			"cal": "112.0"
		},
		{
			"channel": 4,
			"name": "Solar_W_and_N_With_Bat_4900w",
			"type": "CT",
			"model": "generic",
			"phase": "2",
			"cal": "112"
		},
		{
			"channel": 5,
			"name": "Total_Solar_Produced",
			"type": "CT",
			"model": "generic",
			"phase": "2",
			"cal": "184"
		},
		{
			"channel": 6,
			"name": "Hot_Water_House",
			"type": "CT",
			"model": "generic",
			"phase": "2",
			"cal": "103.0"
		},
		{
			"channel": 7,
			"name": "House_Heating_Cooling",
			"type": "CT",
			"model": "generic",
			"phase": "2",
			"cal": "184"
		},
		{
			"channel": 8,
			"name": "Shed_Power_A",
			"type": "CT",
			"model": "generic",
			"phase": "2",
			"cal": "103"
		},
		{
			"channel": 9,
			"name": "Shed_Power_B",
			"type": "CT",
			"model": "generic",
			"phase": "2",
			"cal": "103"
		},
		{
			"channel": 10,
			"name": "Shed_Rear_Power_Box",
			"type": "CT",
			"model": "generic",
			"phase": "2",
			"cal": "112"
		},
		{
			"channel": 11,
			"name": "House_General",
			"type": "CT",
			"model": "generic",
			"phase": "2",
			"cal": "184"
		},
		{
			"channel": 12,
			"name": "Shed_Lighting",
			"type": "CT",
			"model": "generic",
			"phase": "2",
			"cal": "112"
		},
		{
			"channel": 13,
			"name": "House_Outdoor_Power",
			"type": "CT",
			"model": "generic",
			"phase": "2",
			"cal": "103"
		},
		{
			"channel": 14,
			"name": "Shed_32Amp_Outlet",
			"type": "CT",
			"model": "generic",
			"phase": "2",
			"cal": "112"
		}
	],
	"outputs": [
		{
			"name": "Heating and Cooling",
			"units": "Watts",
			"script": "@7"
		},
		{
			"name": "House General",
			"units": "Watts",
			"script": "@11"
		},
		{
			"name": "House Hot Water",
			"units": "Watts",
			"script": "@6"
		},
		{
			"name": "Import Export",
			"units": "Watts",
			"script": "@1"
		},
		{
			"name": "Mains Voltage",
			"units": "Volts",
			"script": "@0"
		},
		{
			"name": "NBN Outdoor Box",
			"units": "Watts",
			"script": "@13"
		},
		{
			"name": "Self Use",
			"units": "Watts",
			"script": "@5-@1"
		},
		{
			"name": "Shed 32 Amp Outlet",
			"units": "Watts",
			"script": "@14"
		},
		{
			"name": "Shed A Circuit",
			"units": "Watts",
			"script": "@8"
		},
		{
			"name": "Shed B Circuit With Pump",
			"units": "Watts",
			"script": "@9"
		},
		{
			"name": "Shed Lighting",
			"units": "Watts",
			"script": "@12"
		},
		{
			"name": "Shed Rear Box",
			"units": "Watts",
			"script": "@10"
		},
		{
			"name": "Solar Bulk Production",
			"units": "Watts",
			"script": "@5"
		},
		{
			"name": "Solar East Production",
			"units": "Watts",
			"script": "@2"
		},
		{
			"name": "Solar North Production",
			"units": "Watts",
			"script": "@4"
		},
		{
			"name": "Solar West Production",
			"units": "Watts",
			"script": "@3"
		}
	],
	"server": {
		"type": "emoncms",
		"node": "IotaWatt\t",
		"postInterval": 10,
		"bulksend": 1,
		"url": "http://emoncms.org",
		"apikey": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
		"userid": "xxxxxx",
		"outputs": [
			{
				"name": 1,
				"script": "@0",
				"units": ""
			},
			{
				"name": 2,
				"script": "@1",
				"units": ""
			},
			{
				"name": 3,
				"script": "@2",
				"units": ""
			},
			{
				"name": 4,
				"script": "@3",
				"units": ""
			},
			{
				"name": 5,
				"script": "@4",
				"units": ""
			},
			{
				"name": 6,
				"script": "@5",
				"units": ""
			},
			{
				"name": 7,
				"script": "@6",
				"units": ""
			},
			{
				"name": 8,
				"script": "@7",
				"units": ""
			},
			{
				"name": 9,
				"script": "@8",
				"units": ""
			},
			{
				"name": 10,
				"script": "@9",
				"units": ""
			},
			{
				"name": 11,
				"script": "@10",
				"units": ""
			},
			{
				"name": 12,
				"script": "@11",
				"units": ""
			},
			{
				"name": 13,
				"script": "@12",
				"units": ""
			},
			{
				"name": 14,
				"script": "@13",
				"units": ""
			},
			{
				"name": 15,
				"script": "@14",
				"units": ""
			}
		]
	},
	"update": "MAJOR"
}

rosspeel, 2017-12-17:
[image]

rcojoe, 2017-12-17:
This looks like an issue happening on the 13th before upgrading from 21 to 27 on morning of the 14th. It appears to me you are not able to resolve dns addresses.  I see the time update failing, checkUpdate failing and emonService timing out on read which usually means that. If everything is working fine and then this stops, It's usually something upstream.  

If you haven't changed any internet settings, you could try restarting your router and seeing if that resolves the issue. 

Joe
Sent from my iPhone

> On Dec 17, 2017, at 3:44 AM, rosspeel <notifications@github.com> wrote:
> 
> Part of iotamsg.txt file.
> 
> 12/13/17 15:51:51 Resending EmonCMS data.
> 12/13/17 16:03:21 EmonService: POST failed. HTTP code: read Timeout
> 12/13/17 16:03:51 Resending EmonCMS data.
> 12/13/17 16:14:53 EmonService: POST failed. HTTP code: read Timeout
> 12/13/17 16:15:23 Resending EmonCMS data.
> 12/13/17 16:51:04 EmonService: POST failed. HTTP code: read Timeout
> 12/13/17 16:51:34 Resending EmonCMS data.
> 12/13/17 17:03:03 EmonService: POST failed. HTTP code: send payload failed
> 12/13/17 17:03:33 Resending EmonCMS data.
> 12/13/17 17:05:04 timeSync: Failed to get NTPtime.
> 12/13/17 18:14:01 EmonService: POST failed. HTTP code: read Timeout
> 12/13/17 18:14:32 Resending EmonCMS data.
> 12/13/17 18:39:21 EmonService: POST failed. HTTP code: read Timeout
> 12/13/17 18:39:51 Resending EmonCMS data.
> 12/13/17 21:04:11 EmonService: POST failed. HTTP code: read Timeout
> 12/13/17 21:04:41 Resending EmonCMS data.
> 12/13/17 21:08:51 EmonService: POST failed. HTTP code: read Timeout
> 12/13/17 21:09:21 Resending EmonCMS data.
> 12/13/17 22:15:01 EmonService: POST failed. HTTP code: read Timeout
> 12/13/17 22:15:31 Resending EmonCMS data.
> 12/13/17 22:46:41 EmonService: POST failed. HTTP code: read Timeout
> 12/13/17 22:47:11 Resending EmonCMS data.
> 12/14/17 02:10:41 EmonService: POST failed. HTTP code: read Timeout
> 12/14/17 02:11:11 Resending EmonCMS data.
> 12/14/17 04:41:46 EmonService: POST failed. HTTP code: send header failed
> 12/14/17 04:42:16 Resending EmonCMS data.
> 12/14/17 08:08:19 checkUpdate: Invalid response from server.
> 12/14/17 08:22:02 EmonService: POST failed. HTTP code: send payload failed
> 12/14/17 08:22:32 Resending EmonCMS data.
> 12/14/17 08:43:11 EmonService: POST failed. HTTP code: read Timeout
> 12/14/17 08:43:41 Resending EmonCMS data.
> 12/14/17 09:10:41 EmonService: POST failed. HTTP code: read Timeout
> 12/14/17 09:11:11 Resending EmonCMS data.
> 12/14/17 09:22:45 EmonService: POST failed. HTTP code: send header failed
> 12/14/17 09:23:15 Resending EmonCMS data.
> 12/14/17 10:06:11 EmonService: POST failed. HTTP code: read Timeout
> 12/14/17 10:06:41 Resending EmonCMS data.
> 12/14/17 10:08:20 Updater: Update from 02_02_21 to 02_02_27
> 12/14/17 10:09:29 Updater: Update downloaded and signature verified
> 12/14/17 10:09:34 Updater: firmware upgraded to version 02_02_27
> 12/14/17 10:09:34 Firmware updated, restarting.
> 
> ** Restart **
> 
> No clock yet: SD initialized.
> 12/14/17 00:09:42 Real Time Clock is running. Unix time: 1513210182
> 12/14/17 00:09:42 Version: 02_02_27
> 12/14/17 00:09:42 Updater: Installing update files for version 02_02_27
> 12/14/17 00:09:42 Updater: Installing GRAPH.JS
> 12/14/17 00:09:42 Updater: Installing INDEX.HTM
> 12/14/17 00:09:44 Updater: Installing TABLES.TXT
> 12/14/17 00:09:44 Updater: Installing CNFSTYLE.CSS
> 12/14/17 00:09:44 Updater: Installing EDIT.HTM
> 12/14/17 00:09:45 Updater: Installing GRAPH.HTM
> 12/14/17 00:09:45 Updater: Installation complete.
> 12/14/17 00:09:45 Reset reason: Software/System restart
> 12/14/17 00:09:45 Trace: 18,13,14,17,18,13,14,17,18,13,14,17,18,13,14,17,18,13,14,17,18,13,14,17,18,13,14,17,18,13,14,15
> 12/14/17 00:09:45 ESP8266 ChipID:416947
> 12/14/17 10:09:45 device name: IotaWatt, version: 3
[... 1133 lines cut ...]
> 12/15/17 17:17:51 EmonService: Resending EmonCMS data:5
> 12/15/17 17:22:53 EmonService: Resending EmonCMS data:6
> 12/15/17 17:28:55 EmonService: Resending EmonCMS data:7
> 12/15/17 17:35:57 EmonService: Resending EmonCMS data:8
> 12/15/17 17:43:59 EmonService: Resending EmonCMS data:9
> 12/15/17 17:53:01 EmonService: Resending EmonCMS data:10
> 12/15/17 17:53:03 EmonService: Unable to post to Emoncms after 10 retries. Restarting ESP.
> 
> ** Restart **
> 
> No clock yet: SD initialized.
> 12/15/17 07:53:05 Real Time Clock is running. Unix time: 1513324385
> 12/15/17 07:53:05 Version: 02_02_27
> 12/15/17 07:53:05 Reset reason: Software/System restart
> 12/15/17 07:53:05 Trace: 99,12,13,14,15,16,17,18,13,14,15,16,17,18,13,14,15,30,37,38,148,149,140,142,143,144,145,148,149,146,147,39
> 12/15/17 07:53:05 ESP8266 ChipID:416947
> 12/15/17 17:53:05 device name: IotaWatt, version: 3
> 12/15/17 17:53:05 Local time zone: 10
> 12/15/17 17:53:08 MDNS responder started
> 12/15/17 17:53:08 You can now connect to http://IotaWatt.local
> 12/15/17 17:53:08 HTTP server started
> 12/15/17 17:53:08 dataLog: service started.
> 12/15/17 17:53:09 dataLog: Last log entry:1513324380
> 12/15/17 17:53:09 statService: started.
> 12/15/17 17:53:09 timeSync

boblemaire, 2017-12-17:
I loaded your config into a test IoTaWatt and was able to reproduce the problem of no feeds in the graph app.  The problem seems to be that the list of feeds that IoTaWatt wants to send back to the graph app is too large and isn't being sent.  At this point, I don't have a lot more to tell you except that I will continue to research this and try to figure out a solution.

version 02_02_27 uses a current version of the ESP8266/Arduino core, which has many changes. Apparently (because you say it was working before the upgrade, I haven't tried the older version yet) there is some new limitation on the size of the response acceptable to the underlying services that IoTaWatt uses.

Your configuration uses very long names, for both the inputs and outputs.  While the IoTaWatt seems to handle them OK, it's not a PV or even a virtual memory machine.  Resources are limited, and at some point I can see that there will need to be some limits placed on the amount of text that can be used.  Perhaps starting with limiting the length of names.  I haven't thought this through yet.

In terms of immediate relief, I had to delete all but  5 of the outputs to get the response message size to be acceptable.  I see that all of the outputs except "Self use" are just the inputs renamed.  Since both the inputs and outputs appear as feeds in graph, I have to ask why the those outputs are needed.

Another problem is that the output names cannot have spaces.  This is not being strictly enforced in the current code, so my bad.  Nevertheless, it's a real limitation that you will need to fix by renaming the outputs that you wish to keep, using underscores as in the input names.

Once I dig in and find the source of the problem in the core, I may be able to issue a fix to send the long response as long as memory holds out.  Otherwise, the long term fix will probably a combination of limiting name lengths and recoding the response handler for feed names to use break up the rtesponse into chunks.

The problem with Emoncms does not seem to be coincident with the new release.  You have posted for many hours since the new release came in, although not without communications problems.  Communication between Emoncms in the UK and Australia is problematic.  I'm working on enhancements, but there are also some other things that are not above suspicion, so we need to keep at this issue to be sure.  In the config that you posted, the node name has been changed to `"node": "IotaWatt\t",`

So I'm not sure if you are or should be seeing different results with the userid removed.  Can you change it back to Peel and update me on the status with Emoncms?

boblemaire, 2017-12-18:
Fixed in current master with PR#113.  Will close when released.

boblemaire, 2018-01-15:
Fix incorporated into release 02_02_28

closed 2018-01-15 by boblemaire