boblemaire/IoTaWatt #127: own compile - no VT active

promy, 2018-02-03:
I tried to compile and flash this version (with and without the changed samplePower.cpp
Ran ok, but, it didn't capture the VT's (and as such no power measurements on the CT's i guess), everything stayed on 0. I changed the sd files (all exept the config) but no luck.

Reflashed back to an old version (02-02-26) i had downloaded (and running on another IotaWatt), and this version worked OK (changed the sd files back since with the new sd files this version showed a wrong "ac cycles sampled/second) (5 vs 28)

boblemaire, 2018-02-04:
Not much to go on here.  Not sure what "this" version means when it is "with and without" one of the components.  All I can say is that the released version 02_02_29 is at auto-update MINOR and running without incident.  Can you download and run the firmware binary for 02_02_29?

boblemaire, 2018-02-18:
No further information after 2 weeks, closing this as insufficient information to reproduce.

closed 2018-02-18 by boblemaire