boblemaire/IoTaWatt #137: Error while trying to build from sources

moose75, 2018-03-04:
Hello !
While trying to build IoTaWatt firmware from unmodified sources under Atom / PlatormIO (Mac OSX 10.3.3), I got the following error :
```
Compiling .pioenvs/iotawatt/lib35f/ESPAsyncTCP_ID305/ESPAsyncTCPbuffer.cpp.o
Compiling .pioenvs/iotawatt/lib35f/ESPAsyncTCP_ID305/SyncClient.cpp.o
Compiling .pioenvs/iotawatt/lib35f/ESPAsyncTCP_ID305/tcp_axtls.c.o
Compiling .pioenvs/iotawatt/libef5/asyncHTTPrequest_ID1836/asyncHTTPrequest.cpp.o

Archiving .pioenvs/iotawatt/lib41c/libCrypto_ID1168.a
In file included from .piolibdeps/asyncHTTPrequest_ID1836/src/asyncHTTPrequest.cpp:1:0:
.piolibdeps/asyncHTTPrequest_ID1836/src/asyncHTTPrequest.h:33:25: fatal error: ESPasyncTCP.h: No such file or directory
 
*********************************************************************
* Looking for ESPasyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPasyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:ESPasyncTCP.h
*
*********************************************************************
 
#include <ESPasyncTCP.h>
^
compilation terminated.

```
Is there any case sensitive problem because "ESPAsyncTCP.h" is writen "ESPasyncTCP.h" in asyncHTTPrequest.h ?

Thanks for your kind help
Greatings from France
Jean-Claude

moose75, 2018-03-06:
Dear All,
As a workaround, I edited ".piolibdeps/asyncHTTPrequest_ID1836/src/asyncHTTPrequest.h" and changed "ESPasyncTCP.h" to "ESPAsyncTCP.h" and I was able to compile and run the firmware.
But please @boblemaire, may you make the correction in the original source file ?
Thanks

boblemaire, 2018-03-07:
Sorry about that.  It does not cause a problem with VScode/pio.  So easy to miss....  
I made the change to asyncHTTPrequest and bumped the version to 1.1.1.  It can take awhile for the PIO crawler to publish the new version.  I can't change the version in lib deps until it's available (I try to stay with strict version control).

closed 2018-03-07 by boblemaire

moose75, 2018-03-07:
No problem for the delayed availability.
Thanks a lot for your great work on that project !

boblemaire, 2018-03-11:
Hey Moose, just a heads up that the platformio crawler has recognized 1.1.1 with the fix you requested.

moose75, 2018-03-12:
I saw the update in PIO, everything works fine now. Thanks Bob !