boblemaire/IoTaWatt #145: Better support for outputs

fredrike, 2018-03-30:
I guess that the aim of the outputs is to follow: https://www.emoncms.org/site/api#feed

I'm missing support for https://iotawatt.local/feed/timevalue.json?id=1 (Last updated time and value for feed)

boblemaire, 2018-04-03:
@fredrike,
The /feed/data API, as well as the /feed/list and a few others, were supported to support the graph application, which is a modified Emoncms graph app.  The version used by IoTaWatt uses a modified feed/data API to better leverage the way data id organized in the datalog.  The graph app has also been modified to make the API requests asynchronous, making the app more responsive.  There is no plan to take this any further.  More likely, the app will be redesigned from the ground up to offer other useful visualizations like solar generation and pie-chart type usage breakdowns.  This is a ways off, as first I'd like to make influx the DB of choice and develop some useful dashboards that can run on the free hosted grafana site.

I don't know much about aRest, but from a glace it looks like something that would need to be compiled in to modify.  That's not how IoTaWatt works, but you are free to fork a branch and have at it.

I take the first comment to be a specific request to add /feed/timevalue to the API list.  Basically, if you can get a response from the IoTaWatt at any particular NTP time, round it down by 5 seconds and that's the time of the last entry in the datalog.  There are no "feeds".  That's an artifact of Emoncms.  IoTaWatt takes a different approach and saves the raw data at 5 second intervals in the datalog.  If it's running, it's logging.

I'm going to close this, but feel free to comment back if you have any questions, or go to the forum to follow up.

closed 2018-04-03 by boblemaire

fredrike, 2018-04-03:
@boblemaire,

Thanks for getting back to me.
It is a bit confusing that we are able to configure "outputs" but that there is no clear documentation on how to access this information (what endpoints are available for accessing the data).

I did take a look at the Status page and found this endpoint http://iotawatt.local/status?inputs=yes&outputs=yes&stats=yes which returns a nicely formatted json with all inputs and outputs 😄.

So after some more investigation I ended up with this configuration for home-assistant.io. I'm reading the config.txt to display the configured names of the outputs.

```yaml
sensor:
  - platform: rest
    name: iotawatt
    json_attributes:
      - inputs
      - outputs
      - stats
    resource: 'http://iotawatt.local/status?inputs=yes&outputs=yes&stats=yes'
    value_template: '{{ value_json.stats.runseconds }}'
    unit_of_measurement: seconds
  - platform: rest
    name: iotawatt config
    json_attributes:
      - inputs
    resource: 'http://iotawatt.local/config.txt'
    value_template: '{{ value_json.device.name }}'

  - platform: template
    sensors:
      iotawatt_in_1:
        unit_of_measurement: W
        friendly_name_template: '{{ states.sensor.iotawatt_config.attributes.inputs[1].name }}'
        value_template: '{{ states.sensor.iotawatt.attributes.inputs[1].Watts }}'
      iotawatt_in_2:
        unit_of_measurement: W
        friendly_name_template: '{{ states.sensor.iotawatt_config.attributes.inputs[2].name }}'
        value_template: '{{ states.sensor.iotawatt.attributes.inputs[2].Watts }}'
      iotawatt_in_3:
        unit_of_measurement: W
        friendly_name_template: '{{ states.sensor.iotawatt_config.attributes.inputs[3].name }}'
        value_template: '{{ states.sensor.iotawatt.attributes.inputs[3].Watts }}'
      iotawatt_in_4:
        unit_of_measurement: W
        friendly_name_template: '{{ states.sensor.iotawatt_config.attributes.inputs[4].name }}'
        value_template: '{{ states.sensor.iotawatt.attributes.inputs[4].Watts }}'
      iotawatt_in_5:
        unit_of_measurement: W
        friendly_name_template: '{{ states.sensor.iotawatt_config.attributes.inputs[5].name }}'
        value_template: '{{ states.sensor.iotawatt.attributes.inputs[5].Watts }}'
      iotawatt_in_6:
        unit_of_measurement: W
        friendly_name_template: '{{ states.sensor.iotawatt_config.attributes.inputs[6].name }}'
        value_template: '{{ states.sensor.iotawatt.attributes.inputs[6].Watts }}'
      iotawatt_in_7:
        unit_of_measurement: W
        friendly_name_template: '{{ states.sensor.iotawatt_config.attributes.inputs[7].name }}'
        value_template: '{{ states.sensor.iotawatt.attributes.inputs[7].Watts }}'
      iotawatt_in_8:
        unit_of_measurement: W
        friendly_name_template: '{{ states.sensor.iotawatt_config.attributes.inputs[8].name }}'
        value_template: '{{ states.sensor.iotawatt.attributes.inputs[8].Watts }}'
      iotawatt_out_0:
        friendly_name_template: '{{ states.sensor.iotawatt.attributes.outputs[0].name }}'
        unit_of_measurement: W
        value_template: '{{ states.sensor.iotawatt.attributes.outputs[0].value }}'
      iotawatt_out_1:
        friendly_name_template: '{{ states.sensor.iotawatt.attributes.outputs[1].name }}'
        unit_of_measurement: W
        value_template: '{{ states.sensor.iotawatt.attributes.outputs[1].value }}'
      iotawatt_out_2:
        friendly_name_template: '{{ states.sensor.iotawatt.attributes.outputs[2].name }}'
        unit_of_measurement: W
        value_template: '{{ states.sensor.iotawatt.attributes.outputs[2].value }}'

``` 

p.s. I'm using this and letting Home Assistant publish my data into influxdb so I can create some nice panels on grafana.

boblemaire, 2018-04-03:
Not familiar with that Fredrike.  I couple of comments though...

There are a couple of issues with using the status feed as a data source.  First, the values are damped so they will display better in the app.  The second thing is that as a source for a time series database like influx or Emoncms, the status is real-time so there is no time stamping or ability to bulk update after an outage.

The latest release, 02_03_04, now in ALPHA, has the new influxDB support.  It will upload missing data after an outage, is timestamped, and is not damped.  The influx interface is documented in the wiki.

fredrike, 2018-04-03:
@boblemaire 

Yes, I'm running `02_03_04` but I don't like the way values are inserted in influxdb 

```
select * from IoT order by time asc limit 10;
time                DM    DM_wH   Livingroom Livingroom_Wh Sovrum_2o3 Sovrum_2o3_wH TM   TM_wH  Total_kWh VVB    VVB_wH Voltage location
----                --    -----   ---------- ------------- ---------- ------------- --   -----  --------- ---    ------ ------- --------
1522418040000000000 0.28  0.0003  0.34       0.0003        588.08     0.5807        0.12 0.0001 0.0013796 807.72 0.7976 217.04  Mjovik

```

Versus:

```
> select time, friendly_name, value from W order by time desc limit 10;
name: W
time                friendly_name                         value
----                -------------                         -----
1522785192755553024 Total                                 1004.623
1522785192748719104 VVB                                   895
1522785192742091008 Total_Wh                              1011.468
1522785192739299072 Vardagsrum                            109
1522785183872164864 Utebelysning Power                    5.3
1522785171003129856 Sparsnäs energy consumption momentary 1825
1522785161887057920 Total                                 1008.03
1522785161880189952 VVB                                   898
1522785161871171840 Total_Wh                              1014.448
1522785156049019136 Sparsnäs energy consumption momentary 1843.2
```