boblemaire/IoTaWatt #167: 3 phase

stef30360, 2018-06-26:
Hi, 
you say
"Here's what one such adapter looks like. These should soon be available from the OEM store, along with various splitters and patch cords needed to connect."
What kind of adapter ? where I can buy this .
regards
SR

boblemaire, 2018-06-27:
I made a batch of prototype adapters and OEM sold them all.  Unfortunately, it is cost prohibitive to produce and sell these because of the high cost to have them tested to UL safety standards.  So I've posted the schematic and PCB layout, so anyone that wants one build one.  OEM has the schematic, and the CE standards for product safety are more reasonable, so perhaps they will decide to build a run of them or offer a DIY thru-hole kit.

I would point out that IoTaWatt has a good track record with derived reference three phase that uses just one VT.  I'd recommend trying that before opting to add two more VTs.

closed 2018-06-27 by boblemaire

stef30360, 2018-06-27:
Ok. No problem, have you got a link  for upload schematic and PCB ?
SR

boblemaire, 2018-06-30:
It's in the hardware directory.