boblemaire/IoTaWatt #172: Cannot upload data to InfluxDB 1.6.1

mabnz, 2018-08-10:
Hi,

Recently I updated to v1.6.1, and it appears to have broken data uploading from iotawatt.

8/10/18 04:58:42z Real Time Clock is running. Unix time 1533877122 
8/10/18 04:58:42z Version 02_03_10

The issue seemed to coincide with a number of crazy RTC updates too:

8/08/18 17:17:59 timeSync: adjusting RTC by -2
2/07/36 18:28:17 timeSync: adjusting RTC by 552240606
8/09/18 03:18:21 timeSync: adjusting RTC by -552240607
2/07/36 18:28:16 timeSync: adjusting RTC by 552190147
8/09/18 17:19:08 timeSync: adjusting RTC by -552190148
8/10/18 14:19:14 timeSync: adjusting RTC by -2

Has anyone else seen the same thing?

boblemaire, 2018-08-11:
I just noticed this issue posting.  For some reason I didn't get a notification.  It's best to report issues with IoTaWatt in the [forum](https://community.iotawatt.com/).  I'm interested in how this happened.  I recently put out a new ALPHA release with updates to the NTP time synchronization code that, while intended to eliminate these kinds of problems, may have caused your problem in some way.  I'd like to get a look at your message log for August 9 when this happened.  Could you post it into the forum please?
I'll leave this open until I verify it is from before the new release, or I find and fix a new problem. 
Thanks.

boblemaire, 2018-08-11:
Verified this is the problem fixed in 02_03_12.

closed 2018-08-11 by boblemaire