boblemaire/IoTaWatt #196: InfluxDB UDP input support

kylehendricks, 2019-01-28:
(Just installed my Iotawatt yesterday and am loving it so far)

I noticed that Iotawatt uses InfluxDB's HTTP API, have you considered adding support for the [InfluxDB UDP Input](https://docs.influxdata.com/influxdb/v1.7/supported_protocols/udp/)?  I used this for ESP32-based house water pressure sensor and it was super easy to implement with very low network overhead.

If this was ever implemented, would it be possible to send data to InfluxDB at a shorter interval?

kylehendricks, 2019-02-03:
Was reading through the forums and saw [this](https://community.iotawatt.com/t/best-practices-for-influxdb/524/8?u=khendricks) post from @boblemaire.

>That said, the actual data posting to influx is a very verbose command line interface and you can dim the lights a little sending a lot of measurements very often

Seems like this could possibly help with that a bit?

boblemaire, 2019-02-03:
I'm not seeing the benefits here.  Correct me if I'm wrong, but this just looks like a different transport protocol (instead of HTTP).  Granted HTTP has more overhead with establishing a connection and headers, but the additional size is trivial compared to the payload.  The real show stopper for me is that it's asynchronous, so I don't know if a packet arrives.  One of the major features of IoTaWatt is that it guarantees delivery.

The real fix for the verbose line protocol is to use compression.  There is a ton of repetition in the postings, so it will compress well.  Moreover, the compression algorithm allows for various levels of compression so IoTaWatt could simply target the low hanging fruit of known repetitive measurement names,  keys, values, time prefixes, etc.  It's on the list.

You can do 5 second posting to influx now.  More often than that has diminishing significance as the increased posting will reduce the sampling.

Finally, this is the kind of discussion I would prefer to do on the forum, so I'm going to close this as an issue.  Thanks for the contribution.

closed 2019-02-03 by boblemaire