boblemaire/IoTaWatt #197: undefined reference to `xbuf::readString()'

kylehendricks, 2019-02-03:
Just tried to build master and am getting a linker error.

It looks like xbuf::readString() from your [asyncHTTPrequest](https://github.com/boblemaire/asyncHTTPrequest) library is declared but not defined.

```
Linking .pioenvs/iotawatt/firmware.elf
.pioenvs/iotawatt/src/samplePower.cpp.o:(.text._Z11samplePhasehht+0x2c): undefined reference to `xbuf::readString()'
.pioenvs/iotawatt/src/samplePower.cpp.o: In function `samplePhase(unsigned char, unsigned char, unsigned short)':
samplePower.cpp:(.text._Z11samplePhasehht+0x5a1): undefined reference to `xbuf::readString()'
collect2: error: ld returned 1 exit status
*** [.pioenvs/iotawatt/firmware.elf] Error 1
```

boblemaire, 2019-02-04:
Are you using PIO to build?  The platformio.ini specifies a library dependency 0f 1.1.9 for asyncHTTPrequest.  I just checked and xbuf::readString() is defined in xbuff.cpp as of 1.1.9.

kylehendricks, 2019-02-04:
I'm using vscode with the PlatformIO plugin.  As far as I can tell it has pulled down asyncHTTPrequest 1.1.9.  I looked at `.piolibdeps/asyncHTTPrequest_ID1836/library.json` and it definitely says version 1.1.9.  I then opened up `xbuf.cpp` and only see a `readString` definition for the one that takes an endPos:
```cpp
String      xbuf::readString(int endPos)
```
Also seeing it here:
https://github.com/boblemaire/asyncHTTPrequest/blob/master/src/xbuf.cpp#L192-L207

boblemaire, 2019-02-04:
Sorry about that.  I guess I stuck that in my local copy during a bleary eyed debug session and never rolled it up to the asyncHTTPrequest git.  I've updated asyncHTTPrequest and bumped the version to 1.1.10.  It usually takes the PIO crwler about 24 hours to scrape a new version from github and make it available.  In the meantime, you can just change the declaration in xbuf.h in your .piolibdeps to include the definition:
`String readString(){return readString(available());}`
That's what the new version does.
Could you close this out when you get 1.1.10 and verify it works for you?  Thanks.