boblemaire/IoTaWatt #204: Use NTP server(s) specified by DHCP

aceat64, 2019-03-12:
I'd like to put my IoTaWatt on a secured vlan with no access to the internet, but currently it requires access to an NTP server in order to ensure good timekeeping. If it can use DHCP Option 42, I could use my local NTP server instead of having to craft firewall rules. This could also be useful for "off-grid" installs.

Optionally, if there was a way to simply change the the NTP server that would work as well.

boblemaire, 2019-03-29:
Quick look isn't promising for getting at DHCP options.  That's all at a lower level.  I can add the ability to specify the URL (or IP?) of the NTP server(s), but I will have to think on how to do that best.  I'm trying to keep the UI simple - not present a lot of options that most folks don't understand.  I'm also trying to avoid users editing the config file as blowing the json syntax causes a lot of headache.  Thinking of adding a library where files can be added with these kinds of specifications.   Bottom line is this will be at least a few months before it gets to a release.  In the meantime, if so inclined, the name is hard-coded in timeservices.cpp so you could compile your own version of the firmware.

Tuckie, 2019-06-25:
Just voicing my support for an option to set the NTP server. I have a similar setup where the device does not have direct access to the internet.

starwolf73, 2019-09-03:
Also voicing my support as all my IoT is on secure VLAN with no access to internet.

boblemaire, 2019-12-05:
I do not see a way to get that from the ESP8266/arduino core services.  If you know how to get the DHCP NTP name, please advise, otherwise will close this issue as unattainable.

Tuckie, 2019-12-23:
What about just being able to specify the NTP server?

Jalle19, 2020-01-07:
You can probably work around it with NAT (e.g. https://www.google.com/amp/s/www.linuxincluded.com/ntp-server-ip-blacklisted-nat-redirection-ftw/amp/)

kockockockoc, 2020-01-26:
@boblemaire, looks like this thread about esp8266/Arduino and "NTP servers over DHCP": https://github.com/esp8266/Arduino/issues/5681

closed 2020-11-01 by boblemaire

ccutrer, 2021-12-21:
Another voice for at least manually specifying the NTP server. That seems much more attainable than getting at the DHCP option.