boblemaire/IoTaWatt #251: Feature request enter ref voltage manually

i00, 2020-01-06:
The voltage at my house varies between 243 - 245 (been using IoTaWatt for about a week) ... I don't really think that you should have to have a separate ref transformer for this ...
I think there should be a way to enter the value manually - if you have a good supply it should vary very little anyway!

ogiewon, 2020-01-06:
I believe, in addition to voltage, the phase angle of the AC waveform is being measured and is critical to accurate power measurement of all of the CT inputs.

boblemaire, 2020-01-17:
@ogiewon right. While you may think your voltage is only varying between 243 and 245, it is actrually varying between -345 and +345 every 20ms in a sine wave.  IoTaWatt measures this about 700 times during that 20ms along with the current at that time.  These 700 measurements are used to developp real power - what you pay for.  There are energy monitors that only measure the current and multiple by a constant nominal voltage.  What they report is apparent power which is typically greater than real power.  I'm closing this out.

closed 2020-01-17 by boblemaire