boblemaire/IoTaWatt #252: iso seconds query error

richardmh, 2020-01-12:
It would be great to have a generic web service exporter which could POST stuff to a node-red endpoint (and a companion generic GET for the last date to keep things in sync) but in the meantime I shall make something which queries my iotawatts internally in the lan and posts the result to my external node-red & db.

However I found a problem which will complicate my sync which seems quite repeatable in version 02_05_02

this works with a 200 response:
http://192.168.0.25/query?format=json&header=yes&resolution=high&missing=null&begin=2020-01-12T06:02&end=s&select=[time.utc.iso,PV_TOTAL.Watts.d1,PV_TOTAL.Wh.d2]&group=5m

... but if you add seconds as per iso spec and your documentation: 
begin=2020-01-12T06:02:00 
you get a 400 response and {"error":"invalid query. unspecified"}

Regards

Richard

boblemaire, 2020-01-17:
Identified and fixed.  Will be corrected in next release after 02_05_02.

richardmh, 2020-01-17:
Thankyou.

A great product.

Richard

boblemaire, 2020-02-12:
Fixed in 02_05_03

closed 2020-02-12 by boblemaire