boblemaire/IoTaWatt #273: Is it possible to use this front end with different use case with modification?

happytm, 2020-05-31:
This is perfect front end for it's intended use case of energy monitor.

Is it possible to use this front end with different use case with modification? I wanted to use it with a ESP8266 based multi sensor device whick will out put about 8 different sensor values. All of them are integers except battery voltage which is float value. I like to collect all these sensor data every few minutes which can be displayed later on in graph or table format. My main goalis to keep data local so I do not have to rely on cloud or internet.

Please guide me.

Thanks.

boblemaire, 2020-06-01:
From the beginning it was conceived that other metrics might be collected, and some of the ones that you mention.  I definitely think it's possible to do, but there are limitations:

- 
- Adding other types of inputs requires that the hardware support the sensors.  The ESP8266 does not have a lot of GPIO pins.  There are no unused pins in the current device.  The pins used for SPI also serve the SDcard, so can't eliminate that.  The I2C also services the RTC, so can't eliminate that.  There are the CS pins for the ADCs if you want to abandon power sampling, and the single ESP8266 ADC is also available.
- 
- Adding different sensor types would require a rethinking of the UI and how to configure.
- 
- MQTT support would need to be added, which is puts pressure on available HEAP.  Another option is to use a HTTP agent to send the MQTT.

If you decide to do this, and want to have a portal to ask questions, please use the "Homebrew" category  in the [iotawatt forum](https://community.iotawatt.com/).

closed 2020-06-01 by boblemaire

happytm, 2020-06-02:
Thanks for your detailed response. I agree there is limitation regarding available gpio pins. I was planning to use  BME280 sensor for Temperaure, Humidity and Pressure and use either APDS-9960 for light or gesture sensor. I was planning to use Arduino Pro Mini as a I2C slave to extend the gpio pins. If ever you decide to implement this framework for ESP32 please consider other use cases. this is very beautiful front end and could be very flexible for other use cases.

Thank anyways for your help.