boblemaire/IoTaWatt #287: Change Content-Type from text/json to application/json for /status

gtdiehl, 2020-12-10:
Retrieving the data from `/status` with any query string, the Content-Type returned is `text/json`.

This doesn't cause any problems, but when I retrieve this page (for example through a python http library) and determine if the page is json or not the function fails as it is expecting `application/json`

Here is an example tcpflow output showing the Content-Type
```
010.000.010.204.58249-010.000.020.202.00080: GET /status?stats HTTP/1.1
Host: 10.0.20.202
Connection: keep-alive
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 11_0_1) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
Accept-Encoding: gzip, deflate
Accept-Language: en-US,en;q=0.9

010.000.020.202.00080-010.000.010.204.58249: HTTP/1.1 200 OK
Content-Type: text/json
Content-Length: 190
Connection: close
```

boblemaire, 2021-02-12:
Just noticed this, sorry.  I'll change it in the next release.

boblemaire, 2021-03-10:
This has been done and will come out shortly in release 02_06_00.  Thanks.

closed 2021-03-10 by boblemaire