boblemaire/IoTaWatt #288: Influx DB Password stored in plaintext and accessible

gtdiehl, 2020-12-11:
I noticed that if I access the '/config.txt' page on an IoTaWatt that has InfluxDB configured the password is shown in plaintext.

I have enabled authentication on the device (not sure if that would protect the page) but wanted to let you know about this possible security issue.

boblemaire, 2020-12-11:
Yes, it is.  It's also sent more or less in plaintext (base64) in every post. IMO that's a bigger issue than the config file.  To the extent it's all on the LAN, that's your security bubble.  If your influx resides on the WAN, then to be secure you would need to use a reverse proxy like nginx to send it out using TLS.

The IoTaWatt authentication is a little better.  It stores only the md5 digest of the password so authenticates without the actual password or transmitting the plaintext password in a message.  Access to the config file is restricted with authentication.  But you're right, it's only as secure as the LAN.

The firmware is open source, so obfuscation is not going to protect it at all.  I don't know of any way to encrypt, and subsequently decrypt it, without having a key that is unknown to others.

gtdiehl, 2020-12-11:
Thank you for the insight!  I didn't realize the password would be sent in cleartext.  Nice to know enabling authentication on the IoTaWatt device will restrict access to the config file.

Closing this issue.

closed 2020-12-11 by gtdiehl