boblemaire/IoTaWatt #302: No "save" button on influxdb v2 uploader page.

ethaniel, 2021-07-08:
Please see screenshot:
[image]

iamphilrob, 2021-07-08:
Just checked my Influx V1 settings (hopefully V2 is similar) 

I think you need to have something in the measurement box - I use iotawatt (use a different name for each iotawatt if you have more than one) - this is one of the key fields in influx that lets you filter/query measurements for reporting

In the field-key box I have $value

Once those are populated you should be able to add tags and measurements and save.

If you are going to upload from the start of 2020 you will have to be very patient - it took 5 days to upload my data (but it will depend on how many measurements you are uploading)

ethaniel, 2021-07-08:
I've been trying to create an output, but don't have a save button as well:
[image]
Latest Chrome: Version 91.0.4472.114 (Official Build) (x86_64)

ethaniel, 2021-07-08:
Ideally I'd love to have all inputs automatically export into influxdb without me having to manually "wire" them through.

ethaniel, 2021-07-08:
Solved both issues, because measurements/outputs don't accept a space in the "Name" field.
I feel like there needs to be a warning, so I am keeping this ticket open.
[image]

boblemaire, 2021-07-15:
This issue is better addressed in the forum.  if you simply searched influx save button there, you would find at least two responses that would have resolved your problem.  I'm closing this issue.

closed 2021-07-15 by boblemaire