boblemaire/IoTaWatt #303: API query not accruing correctly when using min/max

PhilRW, 2021-08-06:
I have two outputs which I use to convert net metering into two separate values: grid energy consumed and energy returned to grid. Here's an example:

net: `subpanel_1 + subpanel_2 + pv`
grid_consumption: `(( subpanel_1 + subpanel_2 + pv) max 0) abs`
grid_return: `(( subpanel1+ subpanel_2 + pv) min 0) abs`

This query http://iotawatt/query?select=[net.wh,grid_consumption.wh,grid_return.wh]&begin=d&end=s&group=all
returns `[[18600,18600,0]]` even though the internal graph+ utility charts it properly. I would expect it to return something more along the lines of  `[[18600,24600,6000]]`

graph+ shows it properly:

[image]

Glazzanz, 2021-08-10:
@PhilRW agree with you, I am also having this issue.

boblemaire, 2021-08-10:
As explained in this forum post [https://community.iotawatt.com/t/solar-import-export/3175](https://community.iotawatt.com/t/solar-import-export/3175), the query above uses group=all, which returns the net value over the period. (Some folks who have true net metering want to know that)  When graphing that function with Graph+, you are implicitly querying the net for two-minute periods.

closed 2021-08-10 by boblemaire