boblemaire/IoTaWatt #322: feature: per-measurement tags for influx

ccutrer, 2021-12-29:
I'd like to be able to add a "type=total" or "type=device" etc. tag to individual measurements sent to Influx, so that in flux I can build a query that automatically excludes one or the other (total consumption vs. individual circuit). this would be especially useful since I'm going to have 4 iotawatts, and to get the complete consumption for the house, influx will need to sum the "total" measurement for all devices.

boblemaire, 2022-01-26:
Adding the ability to specify tags at the individual measurement level is a slippery slope. How many should be allowed?

This is also very complicated from the perspective of adding the spec to the config.txt file.  Currently, measurements are handled as "tagsets" which are also used for: outputs, integrators, emoncms and PVoutput Scripts.

I'd suggest that you can accomplish this by encoding a "total" indicator in the measurement name and using regular expressions in your queries to recognize them.

closed 2022-01-26 by boblemaire

Jalle19, 2023-09-06:
@ccutrer did you come up with a solution for this? I'm faced with the same dilemma.

boblemaire, 2023-09-06:
@Jalle19, This is an old issue but looking at it again, I have the same reservations about measurement level tags.  But it occurs to me that you could adopt a measurement naming convention that could accomplish the same thing using regular expressions.

ccutrer, 2023-09-06:
No. I stopped using InfluxDB when 3.0 came out this spring, and there's not an OSS version. Couple that with my constant issues with high CPU usage for hours and it's seeming inability to reclaim disk space, and I was done with it. I now use TimescaleDB. I plan to write a separate script that will run on my TimescaleDB server to pull the data from IoTaWatt via API.