boblemaire/IoTaWatt #324: Invalid EmonCMS API call

dakaix, 2022-02-19:
emoncms_uploader::handle_query_s() is making an invalid call to the EmonCMS API. 

The EmonCMS  "/input/get" endpoint only accepts GET queries - yet IoTaWatt is sending a POST with no arguments. See https://emoncms.org/site/api#input

EmonCMS replies to this invalid query with the HTML of the home page (45K in size).

**IoTaWatt Query**
```
POST /input/get?node=IotaWatt HTTP/1.1
host:10.10.101.157:80
content-type:application/x-www-form-urlencoded
Authorization:Bearer EmonCMS_Read/Write_API_Key_Here
Content-Length:0
```

**EmonCMS Response**
```
HTTP/1.1 200 OK
Date: Sat, 19 Feb 2022 19:50:52 GMT
Server: Apache/2.4.6 (CentOS) mpm-itk/2.4.7-04 PHP/7.4.24
X-Powered-By: PHP/7.4.24
Expires: Thu, 19 Nov 1981 08:52:00 GMT
Cache-Control: no-store, no-cache, must-revalidate
Pragma: no-cache
Set-Cookie: EMONCMS_SESSID=0f71oe9aam039ev5uacv1g51ho; path=/; HttpOnly; SameSite=Strict
Transfer-Encoding: chunked
Content-Type: text/html; charset=UTF-8

2000
<!doctype html>
<html class="theme-blue sidebar-dark">
<head>
    <meta http-equiv="content-type" content="text/html; charset=UTF-8">
    <meta name="viewport" content="width=device-width, initial-scale=1.0, minimum-scale=1">
    <title>Emoncms - user login </title>

... Output Truncated ...
```

dakaix, 2022-02-27:
I went through the EmonCMS logs and authentication code, and was able to find the difference between the working test queries I had done and the POST to /input/get.

The root cause was the “HTTP_AUTHENTICATION” header not being passed through from Apache to PHP (“Authorisation: Bearer APIKEY”).

This is disabled by default (as a security measure) but EmonCMS tries to enable it via htaccess. While other parts of that .htaccess were working, the specific Authorization statement wasn’t. I have now instead configured it in the main Apache Virtualhost configuration, and after retesting I can confirm that the authenetication header is being passed through - and both EmonCMS and IoTaWatt are working.

This was an unfortunate combination of factors since other authentication sources (those not using the “Authorization” header) were working, such as the “apikey” variable set via GET or POST. If the goal for IoTaWatt is to try and bulletproof this integration it would be worth considering whether to deprecate use of the authentication method as it isn’t enabled in Apache by default, and as I’ve found the supplied EmonCMS htaccess doesn’t always work. The easiest route would be for IoTaWatt to use the “apikey” variable instead, and would be straightforward to implement.

emoncms.log:
```
2022-02-19 19:50:52.021|ERROR|index.php|Not Authenticated|input/get
2022-02-19 19:54:01.581|ERROR|index.php|Not Authenticated|input/get
2022-02-19 19:54:06.466|ERROR|index.php|Not Authenticated|input/get
2022-02-19 19:54:11.379|ERROR|index.php|Not Authenticated|input/get
2022-02-19 20:02:44.001|ERROR|index.php|Not Authenticated|input/get
2022-02-19 20:08:35.998|ERROR|index.php|Not Authenticated|input/get
```

EmonCMS Authentication:
```
$apikey = false;
$devicekey = false;
if (isset($_GET['apikey'])) {
    $apikey = $_GET['apikey'];
} elseif (isset($_POST['apikey'])) {
    $apikey = $_POST['apikey'];
} elseif (isset($_GET['devicekey'])) {
    $devicekey = $_GET['devicekey'];
} elseif (isset($_POST['devicekey'])) {
    $devicekey = $_POST['devicekey'];
} elseif (isset($_SERVER["HTTP_AUTHORIZATION"])) {
    // Support passing apikey on Authorization header per rfc6750, like example:
    //      GET /resource HTTP/1.1
    //      Host: server.example.com
    //      Authorization: Bearer THE_API_KEY_HERE
    
    if (isset($_SERVER["CONTENT_TYPE"]) && $_SERVER["CONTENT_TYPE"]=="aes128cbc") {
        // If content_type is AES128CBC
    } else {
        $apikey = str_replace('Bearer ', '', $_SERVER["HTTP_AUTHORIZATION"]);
    }
}
```

EmonCMS Standard .htaccess:
```
# ------------------------------------------------------------------------------
# Rewrite auth header. Fix Authorization: Bearer
# https://cweiske.de/tagebuch/php-apache-authorization.htm
# ------------------------------------------------------------------------------
<IfModule mod_setenvif.c>
  SetEnvIf Authorization "(.*)" HTTP_AUTHORIZATION=$1
</IfModule>
```

boblemaire, 2022-05-04:
Closing as Emoncms/Apache problem.