boblemaire/IoTaWatt #330: Multiple units on one set of VTs

fukuimaunicorn, 2022-08-03:
Hi there,

I'm sorry to ask here, but I can't find anything relating to this in the docs.
I plan to install multiple IoTaWatt Units. I want the measure my power precisely and therefor use a VT for every phase. Now my question is, if I install one set of VTs (one for each phase) can I connect them to multiple IoTaWatt units as reference. Cause as far as I can tell, the units don't draw any current from the reference and therefor there shouldn't be any problem right?
Because 3 VTs per unit would be a _bit_ excessive..

Hope you can clear this up for me, thanks ^^
- Aron

promy, 2022-08-03:
I'm using 1 set of 3 VT's to give a reference to 5 iotaWatt. I just use a splitter (look for the one's for security camera's)

boblemaire, 2022-08-09:
Promy is right.  You can use CCTV 5.5mm splitters to share VTs.  The use of CCTV splitters is strongly recommended as it is critical that polarity be observed and the same for all units.  Also, when sharing VTs, It's critical to have an independent USB power supply for each unit.  Ground loops and shorts can be set up that lead to inaccurate measurements at best and damaged components at worst.

BTW, the [forum ](https://docs.iotawatt.com/en/master/)is the best place for these types of inquiries.

closed 2022-08-09 by boblemaire