boblemaire/IoTaWatt #335: Second IotaWatt added, first looses all settings and data

lyricnz, 2022-09-27:
I had an Iotawatt setup with default device-name, and had configured the inputs etc, and it was all working fine.

I bought a second device, configured the wifi, and the first unit appears to have lost all it's input settings (inputs, calibration, etc)

Any idea what happened?  This seems like a case where it would be useful to be able to export/import settings for safe storage.

lyricnz, 2022-09-27:
(Maybe unrelated, but I can't get the device reporting proper values - they seem to be about double what they should be!). Unless there's crazy consumption - coincidence?

[image]

lyricnz, 2022-09-27:
Also, after the new device/integration is added, the existing graphs are now a "random" mixture of 1st and 2nd iotawatt, so whatever is used to store the dashboard graph config isn't stable when new iotawatts are added:

[image]

The original iotawatt was: 
main
bungalow
walloven
power1
power2
lights
hotwater

The new iotawatt (currently sitting on a bench) is
grid_a
grid_b
grid_c
solar_a
solar_b
solar_c
shed1
shed2
shed3
shed4
shed5
shed6
shed7

lyricnz, 2022-09-27:
Maybe related - Home Assistant issues with identifying renamed iotawatt integrations, and "corrupted" graphs when second iotawatt is added.  https://github.com/home-assistant/core/issues/79142

boblemaire, 2022-09-30:
You can download the config.txt with the file manager.  If you want to upload it again, rename it to config+1.txt.

When you have two IoTaWatt by the same name, there is the possibility that the browser app loaded the default config from one and then subsequently uploaded it to the other.  There are a lot of variables that would determine how iotawatt.local is resolved at any specific time. Hopefully you have renamed one of them and this will not be an ongoing problem.

closed 2022-09-30 by boblemaire

lyricnz, 2022-09-30:
Is there any way to detect if there's an existing iotawatt.local before the new device is registered with that name?  Then it could add unique suffix etc

lyricnz, 2022-09-30:
FWIW I have renamed both of them, so if I ever add a third, I won't experience this.