boblemaire/IoTaWatt #336: Old espressif8266 version breaks wifi

rspenc29, 2023-01-12:
What's the reason for this and what needs to be done to fix it? 

https://github.com/boblemaire/IoTaWatt/commit/d7621b8b0ec4e60dbad1c358d728619f0c7d2697

Wifi doesn't connect in this version if there are spaces in the ssid and my access points are full on ssid's so I can't just create a new one and I have way too many devices on there to just rename it. I could just change the espressif8266 version and recompile it but seeing as I just got my IotaWatt and haven't installed it yet, I am not familiar with the user interface and probably wouldn't recognize if something was broken.

boblemaire, 2023-01-24:
It's up to you.  I had to make a few changes to MDNS to revert back.  If you use a later core, you will need to troubleshoot any incompatibilities.

rspenc29, 2023-01-24:
Understood. But not being able to connect to wifi is still a bug. I bought a brand new device and if I wasn't experienced with the ESP8266 as most people aren't then I would have probably been returning my device because it don't work.

boblemaire, 2023-01-27:
Unable to reproduce this with any current release. Being unable to connect to WiFi is a common problem that is almost always caused by configuration issues in the user's WiFi.