boblemaire/IoTaWatt #339: Graph+ last 3 hours and last 6 hours has error

chriszylstra, 2023-01-28:
Hi there,

I'm getting this error message I try to select last 3 hours or last 6 hours. 
Note: I do not actually have data from 3 hours ago as I just installed my IoTaWatt unit. Just mentioning  this because the error does not appear on any other timeframe that I do not have data for yet. (such as last 24 hours)

I will wait 6 hours and update this if the issue is still happening or has gone away. It might just be an edge case error of the first 6 hours after installing the device. 

Thanks, 
-cz

[image]

[image]

chriszylstra, 2023-01-29:
Update: Error goes away when there is more than 6 hours of data. So maybe its a non-issue, just letting you know the details.

boblemaire, 2023-06-17:
I'm aware of this issue.  There is a fix in the ESP32 code that needs to be applied to the ESP8266.  But seeing it is an edge issue, doesn't warrant a release.

closed 2023-06-17 by boblemaire