boblemaire/IoTaWatt #345: compact DIN version

gkasprow, 2023-08-03:
Hi
I purchased 3 original IoTaWatt to monitor all my circuits and hardly managed to fit it into my installation box. The audio jacks and 3 VTs took so much space. 
[image]
I need to add a few more DIN rail devices, i.e. individual RCBs so had to optimise the energy monitoring dimensions.
So, I designed compact 3-phase version with embedded voltage transformers and power supply.
I also discovered that in my device the LM358 with 10uF load is oscillating so I fixed that issue by adding compensation network.
[image]
I also used miniature CTs from Talema (AZ-1000) with embedded resistors.
The design is in Altium but one can easily convert it to Kicad when necessary.
The PCB fits nicely low cost D6MG enclosure from GAINTA. No cutouts are needed.
The image shows enclosure with front panel removed.
[image]

The schematic and production files are [here](https://github.com/gkasprow/IoTaWatt_DIN/releases/tag/v1.0rc1)

The sources are [here](https://github.com/gkasprow/IoTaWatt_DIN)

I'm not sure about licensing model. I usually release all my designs using CERN OHL, can you guide me what license shall I use?
Do you think I can use IoTaWatt_DIN name for my design?

6fta, 2023-08-04:
I also started to make a pcb for the rail from ZMPT107-1 transformers, but due to lack of time I did not complete the project.
[image]

gkasprow, 2023-08-04:
I see you added anti-aliasing capacitors at the inputs ?

6fta, 2023-08-05:
Capacitors are optional, I took into account the article in the following link. [ Low current measurement performance of the SCT013](https://www.casler.org/wordpress/low-current-measurement-performance-of-the-sct013/)

gkasprow, 2023-08-05:
I think they are the must. To properly measure signals, one must meet the Nyquist criterium of sampling. For the moment the bandwidth is limited only by CT and VT. For pure sine is fine, but we have broad spectrum of interferences in mains.

gkasprow, 2023-08-05:
One more think. The ADC when sampling, is charging internal capacitor. So it needs low driving impedance. Usually one uses opamp or RC filter where C serves as a reservoir of charge for sampling. In case of CTs, that's not a problem. The VTs have 1k of resistance and this causes measurement error (systematic). Adding 1nF will do two things - limit the BW and meet Nyquist criteria and provide low impedance for ADC.

closed 2023-08-16 by boblemaire

gkasprow, 2023-08-16:
I meant harmonics and EM interferences generated by loads running on frequencies higher than mains, for example, brushed motors, DC/DC converters etc.

perigoso, 2023-11-22:
Here I was thinking about designing a compact DIN rail version, because my cabinet is quite tight...
Good work @gkasprow! I might fabricate one of yours, have you encountered any issues (that I should try to fix before making my own)?

gkasprow, 2023-11-22:
Somebody built it and so far there is only one non-critical issue
https://github.com/gkasprow/IoTaWatt_DIN/issues/1
I still didn't find time to assemble my PCBs. Existing IotaWatt is working well.  Perfect is the enemy of good