boblemaire/IoTaWatt #346: Modbus support

hiperiondev, 2023-09-19:
Excellent project!
Since it could have interesting industrial uses, it would be important to add Modbus access.

pcting, 2024-02-05:
I think something like using this modbus lib might be a good start (https://github.com/emelianov/modbus-esp8266)

for my use case, i have a solaredge inverter with a meter installed. the meter is connected to the inverter through rs485 and the meter has CT clamps reading my mains amps. the solaredge inverter has an ip and supports modbus tcp. i use this endpoint for home assistant integration through the [ha-sunspec](https://github.com/CJNE/ha-sunspec) plugin. 

if iotawatt were to implement metering in the sunspec modbus specification, i can replace the solaredge meter with iotawatt.

pcting, 2024-02-05:
another use case:

this one is an rs485 use case. i have a soyo inverter and it has an a meter that also communicates through rs485 to provide consumption limiting. there's this project here that simulates the meter (https://github.com/syssi/esphome-soyosource-gtn-virtual-meter).

pcting, 2024-02-05:
stepping back from all this, it's probably easier just to build a separate modbus device that polls iotawatt for metrics. but building in modbus support natively would be really nice

boblemaire, 2025-03-30:
Just not possible with the remaining resources in IoTaWatt

closed 2025-03-30 by boblemaire