boblemaire/IoTaWatt #347: Release 02_08_03 connection issues

Antibrumm, 2023-09-29:
Hello

I just realized today that my two Iotawatt units did not communicate anymore with my influx instance. 

After some digging i found that the IPs have changed from the assigned dhcp 192.168… to some random 169.254.. .

I finally went back on the firmware release 02_08_02 and all seems working again. IPs are assigned correctly and the uploder is working.

I did not see anything else broken. The units where still capturing their data. So for now it‘s better to disable auto updates. 

Btw: this is my first issue after a continuous runtime of over 4 years. Really nice work :)

Let me know if I can help with logs or something else. 

Antibrumm

robao23, 2023-12-28:
Hello,

I've come across a peculiar scenario related to this.
Firmware 02_08_03

The device is working fine, but there's a power outage in the entire house, causing the house to lose internet connectivity as well.

Upon power restoration, the IoTaWatt restarts. I'm not certain if the issue arises because it cannot connect to the network immediately after the power is back.

According to the log, it attempts to reconnect every hour. After 9 unsuccessful attempts, the watchdog restarts the IoTaWatt, and it successfully connects on the first attempt.

Is this genuinely a bug? Is there a way to expedite the process so that it resets when it has been disconnected for 10 minutes, or can the reconnection attempt be improved?

Thank you.

`statService: started.
Updater: service started. Auto-update class is MINOR
12/23/23 01:27:57 timeSync: RTC initalized to NTP time
12/23/23 01:27:57 Real Time Clock battery is low.
12/23/23 01:27:57 dataLog: service started.
12/23/23 01:27:57 dataLog: Last log entry 12/23/23 00:49:45
12/23/23 01:27:57 historyLog: service started.
12/23/23 01:27:57 historyLog: Last log entry 12/23/23 00:49:00
12/23/23 01:27:59 Updater: Auto-update is current for class MINOR.

** Restart **

SD initialized.
Real Time Clock not running.
Reset reason: Power on
Trace:  17:38[12], 35:73[131], 65:91[85], 49:86[84], 12:225[40], 21:201[235], 150:184[110], 38:50[154], 126:153[95], 77:188[77], 255:222[254], 24:33[47], 222:138[52], 38:148[128], 191:24[123], 213:240[200], 140:10[106], 3:51[116], 124:111[145], 237:34[65], 212:23[61], 144:210[161], 249:94[165], 180:29[144], 208:186[6], 142:61[77], 126:29[73], 242:87[207], 85:1[170], 38:201[207], 71:22[17], 91:123[207]
ESP8266 ID: 1760914, RTC PCF8523 (68)
IoTaWatt 5.0, Firmware version 02_08_03
SPIFFS mounted.
Local time zone: +1:00
device name: IotaWatt
Connecting with WiFiManager.
RTC not running, waiting for WiFi.
RTC not running, waiting for WiFi.
RTC not running, waiting for WiFi.
RTC not running, waiting for WiFi.
RTC not running, waiting for WiFi.
RTC not running, waiting for WiFi.
RTC not running, waiting for WiFi.
RTC not running, waiting for WiFi.
RTC not running, waiting for WiFi.

** Restart **

SD initialized.
Real Time Clock not running.
Reset reason: Hardware Watchdog
Trace:  11:5, 11:10, 11:15, 11:20, 11:20[1], 11:20[2], 11:20[3], 11:21, 11:22, 11:21, 11:22, 11:20[4], 11:20[5], 11:20[6], 11:20[7], 11:25, 11:30, 11:30[1], 11:30[2], 11:31, 11:31[1], 11:31[2], 11:31[3], 11:30[3], 11:35, 11:36, 11:35, 11:40, 11:45, 11:50, 11:55, 11:70
ESP8266 ID: 1760914, RTC PCF8523 (68)
IoTaWatt 5.0, Firmware version 02_08_03
SPIFFS mounted.
Local time zone: +1:00
device name: IotaWatt
Connecting with WiFiManager.
HTTP server started
WiFi connected. SSID=ok, IP=localIpOk, channel=8, RSSI -54db
timeSync: service started.
statService: started.
Updater: service started. Auto-update class is MINOR
12/28/23 10:28:07 timeSync: RTC initalized to NTP time
12/28/23 10:28:07 Real Time Clock battery is low.
12/28/23 10:28:07 dataLog: service started.
12/28/23 10:28:07 dataLog: Last log entry 12/28/23 01:35:25
12/28/23 10:28:07 historyLog: service started.
12/28/23 10:28:07 historyLog: Last log entry 12/28/23 01:35:00
12/28/23 10:28:08 Updater: Auto-update is current for class MINOR.
`

boblemaire, 2023-12-28:
This issue should be presented in the forum. 

Auto-reconnect is enabled and supposed to be handled by lower level code in the ESP8266.  Unfortunately it doesn't work all that well in this circumstance.  Restarting every 10 minutes would be excessive.

Note that your RTC battery is low so the IoTaWatt doesn't know what time it is until it can connect to the internet.  If you had a good battery, logging would continue without the internet connection so you would not lose any data.

closed 2023-12-28 by boblemaire