boblemaire/IoTaWatt #351: web server UI issue, menu hover state persists on iOS/Safari

tristansokol, 2024-09-02:
Hi! Love the project. 

Was setting up with my phone and noticed that the menu dropdown states persist/are not clearable. On some screens such as `Inputs` you can enter an accessible input option and cancel back out to clear the menu, but for others such as `Outputs` there is nothing visible under the menu to click on and the page is unusable on my phone. 

The easiest fix might be to add a css transition to make the menu expire after a few seconds (20?), even if hovered. Still not a great experience but at least usable. I'm not sure if something better is possible with out reworking how the menu works/if you are interested in updating the menu that much. 

Happy to try a PR if you have a vision for a fix.

dineshpannu, 2025-02-22:
I’m having the same issue. I don’t recall this problem when I first set up the device, so I think a recent iOS version broke it.

Just registering my vote for a fix.

boblemaire, 2025-03-30:
This situation exists on iPhones but you can eliminate the dropdown by simply holding your finger on the background next to it.

closed 2025-03-30 by boblemaire